bulletphysics/bullet3 #632: How long before we should expect bullet 3? in time for blender 2.8?

BluePrintRandom, 2016-05-18:
I would love to have bullet 3 inside the bge in blender 2.8.....

how far off is it?

BluePrintRandom, 2016-06-16:
Bump_Bump

benelot, 2016-09-07:
If you mean the GPU physics pipeline, I think the priorities have changed. It is in an experimental stage and you can test it.

BluePrintRandom, 2016-09-07:
if we can have a GPU based A* + a GPU based AI + gpu physics ragdoll
simulations + GPU armature skinning.....

you could have huge mobs of enemies without any drain from CPU-> gpu
communications.

(think a herd of zombies)

is this possible with Bullet3 + openCL ?

'bullet walking ragdoll crowd simulations' ?

benelot, 2016-09-07:
Well first of all, your GPU has to switch for each of these usages, so you either have lots of GPU cards and use them all at the same time or it would not work. Second, everything with dependencies within the calculation causes large issues on GPU. A* might be impossible to do in full-parallel, except that you can expand a step further everywhere at the same time. GPU based AI sounds unnecessary given the current state of Game AI. Once we have trained Neural Networks that play, we are talking again about this. The GPU physics works, except for large piles of objects (so called islands), they again cause dependencies. 

All in all: We are not there yet from my point of view.

BluePrintRandom, 2016-09-07:
The idea behind a gpu ai was to avoid cycling data from gpu to cpu, and to
go in a order where the gpu already had the info it was dependent on from
the previous calc

like

physics
ai
applyForces from ai for next frame
skin ragdoll and draw world

but these would all need to complete before its time to draw the next
frame....

(a fully gpu bound gameloop)

but I guess a player would have no ability to input anything....

tlf30, 2019-02-08:
@erwincoumans is there a reason opencl support has basically been dropped? It was a feature that I believe many developers were looking forward to.

erwincoumans, 2019-02-08:
Sure, there are no contributions to OpenCL support. My focus is on robotics and reinforcement learning those days, in the form of [PyBullet](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.2ye70wns7io3).
Since it is unlikely anyone is going to have time and resources for OpenCL, let's close this issue for now. If a developer shows up we can re-open.

closed 2019-02-08 by erwincoumans