bulletphysics/bullet3 #643: "syntax error parse error" when running Example Browser compiled with VS2015

gtzampanakis, 2016-05-27:
* Downloaded the latest 2.83.7, 
* created the VS2010 files by running build3/vs2010.bat ,
* opened the *.sln file using VS2015 CE (which caused the projects to be converted to VS2015),
* built the solution,
* ran the App_ExampleBrowser_vs2010_debug.exe file
* received the error:
```

Version = 2.1.0 - Build 8.15.10.2900
Vendor = Intel
Renderer = Intel(R) HD Graphics
Compile failed:
ERROR: 0:3: 'location' : syntax error parse error

Assertion failed: 0, file ..\..\examples\OpenGLWindow\LoadShader.cpp, line 46
```

erwincoumans, 2016-05-27:
Please download/use the latest trunk version. It looks like you need to upgrade your graphics driver, it is some GLSL shader compiler issue. Good luck!

closed 2016-05-27 by erwincoumans

gtzampanakis, 2016-05-27:
Just tried with the trunk version and getting the same error. Upgrading the graphics driver is not an option since I am already using the latest version.

Isn't there another solution to see the examples?

gtzampanakis, 2016-05-27:
I hope that the only thing I will be missing are the examples. Is this correct?

erwincoumans, 2016-05-27:
What version of Intel integrated GPU do you have exactly? What operating system+version?
It looks like OpenGL version 2.1 which is over 10 years old, most Intel GPU drivers support OpenGL version 3+ or 4+.
You can use the --opengl2 command-line option, but most demos/examples won't work, so it is better to fix the OpenGL issue.

gtzampanakis, 2016-05-28:
On Device Manager it is listed: "Intel(R) HD Graphics".

OS: Windows 10 Pro, Version 1511

I tried with the --opengl2 option but I receive the exact same error.

gtzampanakis, 2016-05-28:
It looks like my GPU only supports OpenGL 2: https://www-ssl.intel.com/content/www/us/en/support/graphics-drivers/000005524.html

So I guess I cannot see the examples. I will have to make my own using SDL or something.