bulletphysics/bullet3 #655: Error building on Linux

gtzampanakis, 2016-06-14:
Attempting to build on Linux using g++ 4.4.7 and running the command:

```
./premake4_linux gmake
cd gmake
make config=release32
```

results in:

```
==== Building BulletInverseDynamicsUtils (release32) ====
Creating obj/x32/Release/BulletInverseDynamicsUtils
MultiBodyTreeCreator.cpp
In file included from ../../src/BulletInverseDynamics/IDConfig.hpp:65,
                 from ../../Extras/InverseDynamics/MultiBodyTreeCreator.hpp:6,
                 from ../../Extras/InverseDynamics/MultiBodyTreeCreator.cpp:1:
../../src/BulletInverseDynamics/details/IDLinearMathInterface.hpp: In constructor ‘btInverseDynamicsBullet3::vecx::vecx(int)’:
../../src/BulletInverseDynamics/details/IDLinearMathInterface.hpp:51: error: class ‘btInverseDynamicsBullet3::vecx’ does not have any field named ‘btVectorX’
../../src/BulletInverseDynamics/details/IDLinearMathInterface.hpp: In member function ‘const btInverseDynamicsBullet3::vecx& btInverseDynamicsBullet3::vecx::operator=(const btVectorX<float>&)’:
../../src/BulletInverseDynamics/details/IDLinearMathInterface.hpp:53: error: expected type-specifier before ‘btVectorX’
../../src/BulletInverseDynamics/details/IDLinearMathInterface.hpp:53: error: expected ‘>’ before ‘btVectorX’
../../src/BulletInverseDynamics/details/IDLinearMathInterface.hpp:53: error: expected ‘(’ before ‘btVectorX’
../../src/BulletInverseDynamics/details/IDLinearMathInterface.hpp:53: error: missing template arguments before ‘*’ token
../../src/BulletInverseDynamics/details/IDLinearMathInterface.hpp:53: error: expected primary-expression before ‘>’ token
../../src/BulletInverseDynamics/details/IDLinearMathInterface.hpp:53: error: expected ‘)’ before ‘;’ token
make[1]: *** [obj/x32/Release/BulletInverseDynamicsUtils/MultiBodyTreeCreator.o] Error 1
make: *** [BulletInverseDynamicsUtils] Error 2
```

closed 2016-06-14 by erwincoumans

minghia, 2018-01-17:
Just checked bullet3 out from GitHub and it appears that change ba8964c4ac reversed the fix that was applied for this problem.

minghia, 2018-01-17:
Here is a patch against the head. 
diff --git a/src/BulletInverseDynamics/details/IDLinearMathInterface.hpp b/src/BulletInverseDynamics/details/IDLinearMathInterface.hpp
index 5bb4a33..a44f779 100644
--- a/src/BulletInverseDynamics/details/IDLinearMathInterface.hpp
+++ b/src/BulletInverseDynamics/details/IDLinearMathInterface.hpp
@@ -49,9 +49,9 @@ inline mat33 operator*(const idScalar& s, const mat33& a) { return a * s; }
 
 class vecx : public btVectorX<idScalar> {
 public:
-       vecx(int size) : btVectorX(size) {}
+       vecx(int size) : btVectorX<idScalar>(size) {}
        const vecx& operator=(const btVectorX<idScalar>& rhs) {
-               *static_cast<btVectorX*>(this) = rhs;
+               *static_cast<btVectorX<idScalar>*>(this) = rhs;
                return *this;
        }

minghia, 2018-01-17:
`diff --git a/src/BulletInverseDynamics/details/IDLinearMathInterface.hpp b/src/BulletInverseDynamics/details/IDLinearMathInterface.hpp
index 5bb4a33..a44f779 100644
--- a/src/BulletInverseDynamics/details/IDLinearMathInterface.hpp
+++ b/src/BulletInverseDynamics/details/IDLinearMathInterface.hpp
@@ -49,9 +49,9 @@ inline mat33 operator*(const idScalar& s, const mat33& a) { return a * s; }
 
 class vecx : public btVectorX<idScalar> {
 public:
-       vecx(int size) : btVectorX(size) {}
+       vecx(int size) : btVectorX<idScalar>(size) {}
        const vecx& operator=(const btVectorX<idScalar>& rhs) {
-               *static_cast<btVectorX*>(this) = rhs;
+               *static_cast<btVectorX<idScalar>*>(this) = rhs;
                return *this;
        }
`

minghia, 2018-01-17:
[patch.txt](https://github.com/bulletphysics/bullet3/files/1637764/patch.txt)
Let's try that again.

erwincoumans, 2018-01-17:
Odd, none of our compilers complain about it (Linux gcc + clang, Mac, Windows,etc).
I re-applied it here: https://github.com/bulletphysics/bullet3/pull/1518/commits/c6a350b807f7b715836f57ac115faca2559d3ba7

minghia, 2018-01-17:
I'm using a really old version of gcc version 4.4.7. That is possibly why it picked it up and the g++ compiler you're using possibly has different default c++ version which allows the syntax. It was the least I could do.

PR #656 (2016-06-14): Fixed: Error because of unparameterized btVectorX
Fixes #655