bulletphysics/bullet3 #670: btAxisSweep3 destroyProxy() crash during collision world cleanup

TheChuckster, 2016-06-24:
Any ideas what could be causing this (happens on both iOS and Android)?

`06-23 16:54:10.485 1431-1431/? D/CrashAnrDetector: Build: Verizon/kltevzw/kltevzw:4.4.4/KTU84P/G900VVRU1ANI2:user/release-keys
                                                   Hardware: MSM8974
                                                   Revision: 14
                                                   Bootloader: G900VVRU1ANI2
                                                   Radio: unknown
                                                   Kernel: Linux version 3.4.0-2525521 (dpi@SWDD5816) (gcc version 4.7 (GCC) ) #1 SMP PREEMPT Sun Sep 14 18:04:11 KST 2014
                                                   
                                                   *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
                                                   Build fingerprint: 'Verizon/kltevzw/kltevzw:4.4.4/KTU84P/G900VVRU1ANI2:user/release-keys'
                                                   Revision: '14'
                                                   pid: 27215, tid: 27215, name: rsive.Lingoland  >>> com.LearnImmersive.Lingoland <<<
                                                   signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 00000020
                                                       r0 82509f04  r1 00000000  r2 798f0c58  r3 79583b40
                                                       r4 82509f04  r5 00000000  r6 798f0c58  r7 8848dd70
                                                       r8 00000000  r9 78560c40  sl 7857cd1c  fp bee99164
                                                       ip 837b13b8  sp bee98ad8  lr 784c99af  pc 784c76be  cpsr 200f0030
                                                       d0  206e754a3e38323c  d1  c3b9eb5331203332
                                                       d2  c388d4b941e220b2  d3  4161fb5ec3c979a7
                                                       d4  c3b9eb53c38e8158  d5  41e220b2c3b9eb53
                                                       d6  c3c979a7c388d4b9  d7  c38e81584161fb5e
                                                       d8  c290000000000000  d9  0000000000000000
                                                       d10 0000000000000000  d11 0000000000000000
                                                       d12 0000000000000000  d13 0000000000000000
                                                       d14 0000000000000000  d15 0000000000000000
                                                       d16 433a3a656e656353  d17 6f4370556e61656c
                                                       d18 446e6f6973696c6c  d19 6e6f697463657465
                                                       d20 61656c43203a2928  d21 6c6f63207075206e
                                                       d22 77206e6f6973696c  d23 2e2e2e20646c726f
                                                       d24 3fd1b10853a79915  d25 3fd554fa9ca0567a
                                                       d26 3fdb6db71408e660  d27 3fe33333332d4313
                                                       d28 3ff0000000000000  d29 0000000000000001
                                                       d30 fff0000000000000  d31 ffffffff00000000
                                                       scr 60000012
                                                   
                                                   backtrace:
                                                       #00  pc 001516be  /data/app-lib/com.LearnImmersive.Lingoland-34/libLingolandNativeActivity.so
                                                       #01  pc 001539ab  /data/app-lib/com.LearnImmersive.Lingoland-34/libLingolandNativeActivity.so (btDbvt::remove(btDbvtNode*)+6)
                                                       #02  pc 00147435  /data/app-lib/com.LearnImmersive.Lingoland-34/libLingolandNativeActivity.so (btDbvtBroadphase::destroyProxy(btBroadphaseProxy*, btDispatcher*)+20)
                                                       #03  pc 00096d07  /data/app-lib/com.LearnImmersive.Lingoland-34/libLingolandNativeActivity.so (btAxisSweep3Internal<unsigned int>::destroyProxy(btBroadphaseProxy*, btDispatcher*)+18)
                                                       #04  pc 000b2e9d  /data/app-lib/com.LearnImmersive.Lingoland-34/libLingolandNativeActivity.so (btCollisionWorld::~btCollisionWorld()+72)
                                                   
                                                   stack:
                                                            bee98a98  78560c40  /data/app-lib/com.LearnImmersive.Lingoland-34/libLingolandNativeActivity.so
                                                            bee98a9c  bee98ac8  [stack]
                                                            bee98aa0  798f0c58  
                                                            bee98aa4  784b3deb  /data/app-lib/com.LearnImmersive.Lingoland-34/libLingolandNativeActivity.so (btHashedOverlappingPairCache::processAllOverlappingPairs(btOverlapCallback*, btDispatcher*)+74)
                                                            bee98aa8  784b3da1  /data/app-lib/com.LearnImmersive.Lingoland-34/libLingolandNativeActivity.so (btHashedOverlappingPairCache::processAllOverlappingPairs(btOverlapCallback*, btDispatcher*))
                                                            bee98aac  798f0c58  
                                                            bee98ab0  7d3bef50  
                                                            bee98ab4  88479c60  
                                                            bee98ab8  00000eb2  
                                                            bee98abc  7857cd1c  
                                                            bee98ac0  bee99164  [stack]
                                                            bee98ac4  784b3d97  /data/app-lib/com.LearnImmersive.Lingoland-34/libLingolandNativeActivity.so (btHashedOverlappingPairCache::removeOverlappingPairsContainingProxy(btBroadphaseProxy*, btDispatcher*)+22)
                                                            bee98ac8  7856f550  /data/app-lib/com.LearnImmersive.Lingoland-34/libLingolandNativeActivity.so
                                                            bee98acc  8360dad8  
                                                            bee98ad0  784b3f69  /data/app-lib/com.LearnImmersive.Lingoland-34/libLingolandNativeActivity.so
                                                            bee98ad4  784b3dc7  /data/app-lib/com.LearnImmersive.Lingoland-34/libLingolandNativeActivity.so (btHashedOverlappingPairCache::processAllOverlappingPairs(btOverlapCallback*, btDispatcher*)+38)
                                                       #00  bee98ad8  82509f04  
                                                            bee98adc  00000000  
                                                            bee98ae0  798f0c58  
                                                            bee98ae4  784c99af  /data/app-lib/com.LearnImmersive.Lingoland-34/libLingolandNativeActivity.so (btDbvt::remove(btDbvtNode*)+10)
                                                       #01  bee98ae8  00000000  
                                                            bee98aec  5f766572  /dev/ashmem/dalvik-he
`

Code snippet (I'm almost certain `SetupCollisionDetection()` and `CleanUpCollisionDetection()` do not interleave):

[... 24 lines cut ...]
}

void Scene::CleanUpCollisionDetection() {
    if (bt_collision_world != nullptr) {
        btCollisionObjectArray& collision_objs = bt_collision_world->getCollisionObjectArray();
        if (collision_objs.size() == 0) { LOGW("No collision objects: (collision_objs.size() == 0)"); }
    }

    LOGW("Scene::CleanUpCollisionDetection(): Clean up collision world ...");
    if (bt_collision_world) { delete bt_collision_world; bt_collision_world = nullptr; }
    // delete mSolver;

    LOGW("Scene::CleanUpCollisionDetection(): Clean up broadphase ...");
    if (bt_broadphase)      { delete bt_broadphase; bt_broadphase = nullptr; }

    LOGW("Scene::CleanUpCollisionDetection(): Clean up dispatcher ...");
    if (bt_dispatcher)      { delete bt_dispatcher; bt_dispatcher = nullptr; }

    LOGW("Scene::CleanUpCollisionDetection(): Clean up collision configuration ...");
    if (bt_collision_configuration) { delete bt_collision_configuration; bt_collision_configuration = nullptr; }

    LOGW("Scene::CleanUpCollisionDetection(): SURVIVED cleaning up collision configuration!");
}`

Think somehow something is getting delete'd twice?

erwincoumans, 2016-06-24:
Yes, likely deleted twice. Did you insert the object in the world twice?

Just compare your initialization and destruction with the Bullet examples, 
especially delete stuff in the reverse order of creation:
https://github.com/bulletphysics/bullet3/blob/master/examples/CommonInterfaces/CommonRigidBodyBase.h

erwincoumans, 2016-06-27:
Better use the forums at http://bulletphysics.org, and use previous suggestion to track down the issue.

closed 2016-06-27 by erwincoumans