bulletphysics/bullet3 #681: CollisionShape and btRigidBody UserIndex influences GLRenderer in ExampleBrowser

benelot, 2016-07-02:
While making another example for the bullet example browser, I was using the UserIndex to distinguish objects. After setting the user indices, I started to get strange issues. So I tried if they were occurring in the other examples as well.

Adding a user index to the groundshape (as shown below) in the BasicExample returns a strange error to me:

        ///create a few basic rigid bodies
        btBoxShape* groundShape = createBoxShape(btVector3(btScalar(50.),btScalar(50.),btScalar(50.)));
        groundShape->setUserIndex(1);

The following issue occurs:

> OpenGLWindow/GLInstancingRenderer.cpp:483: virtual int GLInstancingRenderer::registerGraphicsInstance(int, const float*, const float*, const float*, const float*): Assertion `shapeIndex == (m_graphicsInstances.size()-1)' failed.

If you go to the MotorDemo and add the user index to the section which adds damping:

        // Setup some damping on the m_bodies
        for (i = 0; i < BODYPART_COUNT; ++i)
        {
                m_bodies[i]->setDamping(0.05, 0.85);
                m_bodies[i]->setDeactivationTime(0.8);
                //m_bodies[i]->setSleepingThresholds(1.6, 2.5);
                m_bodies[i]->setSleepingThresholds(0.5f, 0.5f);
                m_bodies[i]->setUserIndex(2);
        }

The following issue occurs:

> OpenGLWindow/GLInstancingRenderer.cpp:304: virtual void GLInstancingRenderer::writeSingleInstanceTransformToCPU(const float*, const float*, int): Assertion `srcIndex<m_data->m_totalNumInstances' failed.

I really have no idea why this happens.

erwincoumans, 2016-07-02:
The user index is used in the ExampleBrowser itself at the moment, to link graphics to physics indices.
It would require a little bit of refactoring to remove this use of user index. While introducing the TinyRenderer as an option, it uses its own data structure to maintain this mapping, we could do the same for the OpenGL 'InstancingRenderer'.

erwincoumans, 2016-07-03:
By the way, I'll have a look how much work it is to refactor the examples to not hijack the user index and get back here.

benelot, 2016-07-03:
No problem, at least I know now why some of my issues occurred such as some
objects disappearing from screen. The errors were funny.

closed 2021-04-26 by erwincoumans