bulletphysics/bullet3 #684: How to remove btRigidBody from world in example browser

benelot, 2016-07-08:
I am implementing a bullet example browser demo which involves removing objects during runtime. The physical objects can be removed using removeRigidBody(...) , but the graphical object stays there. How do I remove it?

MrVallentin, 2016-07-09:
I would call this an issue that suits being here. Did you try [Stack Overflow](http://stackoverflow.com/questions/tagged/bulletphysics) or the [Bullet physics forum](http://www.bulletphysics.org/Bullet/phpBB3/)?

erwincoumans, 2016-07-09:
You cannot remove individual graphics instances, there is only the option to remove _all_ graphics instances at the end of the example/program. This simplifies the renderer.
As an alternative, in my upcoming Pull request you can scale down the graphics instance to zero, so it will be invisible.

		int gfxIndex = collisionObject->getUserIndex();
		float scale[3] = {0,0,0};
		m_guiHelper->getRenderInterface()->writeSingleInstanceScaleToCPU(scale,gfxIndex);
		collisionObject->setUserIndex(-1);

Obviously, you will ultimately run out of memory/instances, if you add/remove in a loop, so don't do that.

benelot, 2016-07-11:
@MrValentin: This is pretty much the only place to ask this (apart from private mail), since it is cutting edge code, entirely related to the example browser, making it easiest to ask Erwin. And here is basically Erwin's door to ring.

@Erwin: ok, that means my breakout game example will have to truncate the blocks that are hit in some way you described. Thanks.

erwincoumans, 2017-05-04:
I finally got to implement removal of single graphics instances: https://github.com/erwincoumans/bullet3/commit/152e3da3e4cad74b4519e48cd26b694ca11f8dba

benelot, 2017-05-29:
Sounds great. I just updated my evolution example with it!

closed 2017-05-29 by benelot