bulletphysics/bullet3 #691: setIgnoreCollisionCheck and CCD

sponji, 2016-07-11:
Seems like setIgnoreCollisionCheck doesn't work with CCD. The rigid bodies move through each other when they go slower, but when there's enough motion they collide. Is this a bug or am I just doing something silly?

benelot, 2016-07-11:
I am not sure if I am correct, but I remember having read that CCD only triggers with at a higher velocity than a certain threshold. Therefore, slowly moving objects are not handled with CDD. That is all I know, maybe it helps you already.

sponji, 2016-07-11:
Yeah, that's what I meant with "enough motion", going over the threshold so it actually starts using CCD.

erwincoumans, 2016-08-04:
It is probably a bug,  ccd collisions need to be filterered the same as discrete collisions. A patch is welcome, otherwise I look at it once I find some time.

PR #1662 (2018-05-03): Fixed a collision check bug with CCD
This should fix #691
closed 2018-05-03 by erwincoumans