bulletphysics/bullet3 #692: pybullet renderImage m_depthBuffer is upsidedown

matpalm, 2016-07-11:
at first i thought it'd be as simple as just including `flip_vertically` (as is done for the `m_rgbColorBuffer`) but i see now that `m_depthBuffer` isn't a `TGAImage` so i'm unsure what helper to use to do this flip...

there's a simple workaround though; flipping after the render...
```
_, _, _, depth = pybullet.renderImage(...)
numpy.flipud(depth)
```

erwincoumans, 2016-07-11:
Is that an interview question? Attached is my untested whiteboard implementation, please test :-)

[swap_depth.zip](https://github.com/bulletphysics/bullet3/files/357655/swap_depth.zip)

matpalm, 2016-07-11:
haha. not an interview question. was thinking that would be likely already be available in some helper function i didn't know about, but i guess not then :)

erwincoumans, 2016-07-11:
Heh, the btAlignedObjectArray<> is a small subset of std::vector, no kitchen-sink features.
Does my patch work for you?

matpalm, 2016-07-12:
I check first thing tomorrow!

matpalm, 2016-07-14:
fixed in @6c856d9fa3865d990c9afcad02d4f97fea8aafa6

closed 2016-07-14 by matpalm