bulletphysics/bullet3 #702: instability with small capsule shapes

Mischanix, 2016-07-18:
https://github.com/bulletphysics/bullet3/blob/master/src/BulletCollision/CollisionShapes/btCapsuleShape.cpp#L136

Tiny capsules had too much inertia leading to instability.  I changed this to use `m_collisionMargin` and my life got a lot easier.

erwincoumans, 2016-07-18:
Thanks for the feedback, that is a good point. I'll fix it right now.

cprogrammer1994, 2017-01-25:
Dropping a capsule onto a static box seems to be unstable running a long simulation.

Details:
```
gravity: (0, 0, -10)
capsuleZ: radius = 5 height = 10 origin = (1, 0, 30) mass = 1
box: half_extents = (10, 10, 1) origin = (0, 0, -1) mass = 0 (static)
```

The capsule is resting for a long time.
After 3000 seconds the capsule is gone (accelerates to a random direction).

**This does not happen if the box origin xy is equal to the capsule origin xy**

erwincoumans, 2017-01-25:
@cprogrammer1994  I just made some changes yesterday, related to this. Are you using the very latest github version?a
Do you have a modified demo, such as the BasicDemo in the ExampleBrowser, to reproduce this?
Which operating system, compiler version, build system used (premake or cmake)?

cprogrammer1994, 2017-01-25:
updated my fork in the morning

erwincoumans, 2017-01-25:
Thanks, but I need a reproduction case as a modified C++ example (BasicDemo for example), and the info requested in my previous message. Thanks a lot!

cprogrammer1994, 2017-01-25:
I will make an example in the example browser then submit the snippet here. Thank you for your quick response.

erwincoumans, 2017-01-25:
By the way, capsules will start rolling without rolling friction,  due to persistent contact manifolds: the cached contacts are not exactly located under the center-of-mass, so that capsules (and cylinders etc) start rolling slowly. See the RollingFrictionDemo how to add rolling and spinning friction. Did you add rolling and spinning friction?

cprogrammer1994, 2017-01-25:
hmm, my problem was with the acceleration. When I first noticed this issue it was with a randomly falling capsule that gain energy from nowhere. Friction values are not changed.

This is the initial scene we noticed the acceleration after we thought the capsule was resting:

[image]

erwincoumans, 2017-01-25:
You need some rolling and spinning friction, due to small persistent contact manifold imprecision to prevent rolling.
I'll wait for some modified ExampleBrowser example (attach a full zipped cpp file that compiles out-of-the-box without no external dependencies, not a snippet). Thanks!

cprogrammer1994, 2017-01-25:
Okey

cprogrammer1994, 2017-01-25:
My changes: https://github.com/bulletphysics/bullet3/compare/master...cprogrammer1994:resting-capsule
Only the BasicDemo folder: [BasicDemo.zip](https://github.com/bulletphysics/bullet3/files/730602/BasicDemo.zip)

So this capsule accelerates. What is the best practice when I want to avoid these kind of things?
`DISABLE_DEACTIVATION` is important for us.

erwincoumans, 2017-01-25:
As I mentioned in my previous posts, you need to set rollingFriction and spinningFriction, for example:

body->setRollingFriction(0.1);
body->setSpinningFriction(0.1);

You can tweak that 0.1 number, but too small and it will start rolling.

What part of 'rollingFriction' and 'spinningFriction' is unclear?

See attached.
[BasicExample.zip](https://github.com/bulletphysics/bullet3/files/730635/BasicExample.zip)

cprogrammer1994, 2017-01-25:
Got it! Thank you.

erwincoumans, 2017-01-26:
small capsules should be fine now, using a sufficient small timestep and setRollingFriction/setSpinningFriction etc. Let's close the issue.

closed 2017-01-26 by erwincoumans