bulletphysics/bullet3 #714: Modernise bullet in blender?

BluePrintRandom, 2016-07-23:
Can we get btmultibody and other newer robotics simulation systems in blender and blender game, walking ragdolls for games and crowd simulation will be very handy.

erwincoumans, 2016-07-23:
Can I get a castle, private beach and a few unicorns and tons of Pokemon Go's for my kids?

closed 2016-07-23 by erwincoumans

BluePrintRandom, 2016-07-23:
the blender 2.8 project seems well funded, and bullet advancements would make blender better for everyone.

Maybe a coder could be paid to do it?

erwincoumans, 2016-07-23:
Let's first have walking ragdolls in Bullet. Using TensorFlow and some reinforcement learning this could be done in the future. Once that is done, someone can be paid to integrate it into Blender.
Not now.

BluePrintRandom, 2016-07-24:
I actually already have made walking ragdolls :D

I use Ik - and attempt to match a animation, but I can devaiate the
animation angle, height, and speed, and length

however I can't get enough friction from the feet
https://www.youtube.com/watch?v=BONoJs_0JAY

BluePrintRandom, 2016-07-24:
http://www.pasteall.org/blend/42836 - this is the same system but without
adding force impulses per step to compensate for not being able to scale
friction past 100 on feet and floor.

for blender 2.77

Space = kill ragdoll (start revive timer)

WASD

BluePrintRandom, 2016-07-24:
https://www.youtube.com/watch?v=IUAGqXMivd4