bulletphysics/bullet3 #781: Question: common part of two collided objects

arzezniczak, 2016-09-12:
Hello,
I have a question. Is possible with Bullet engine to get a common part (product) of two collided objects?
Let's say I have two boxes: (0, 0, 0) to (100, 100, 100) and (50, 50, 50) to (150, 150, 150). What I want to know as a result is a box (50, 50, 50) to (100, 100, 100).

ongamex, 2016-09-12:
Can't give you the exact code example right now, but basically you need to create a new world and ask bullet only to compute the manifold points. You will receive your collision callback as usual (in there you can compute the bounding box that you're searching for using the manifold points), but collision response is going to be performed by bullet.

erwincoumans, 2016-09-12:
Generally this would be a boolean operation (intersection), which requires constructive solid geometry engine. Once you take orientation into account, or more complex geometry, this is a non-trivial problem that Bullet doesn't solve, but it would be a useful feature. At the moment, you only get a hand-full of contact points. Internally, the box-box will search for the 2 most co-planar faces, and performs an intersection operation on those two planes. If you want/need it, search for another CSG/boolean operation library for it, there are some open source ones.

closed 2016-09-12 by erwincoumans