bulletphysics/bullet3 #818: error: ‘btintptr_t’ was not declared in this scope for BUILD_EXTRAS

jiapei100, 2016-10-07:
The Error...

`/bullet3/Extras/Serialize/makesdna/makesdna.cpp:1068:9: error: ‘btintptr_t’ was not declared in this scope
`

erwincoumans, 2016-10-07:
It compiles fine on Windows, Linux and Mac OSX, so please provide ALL details, including operating system+version, compiler+version, all build system steps etc.

benelot, 2017-03-19:
Just had this on Linux as well. I could compile it recently, but not anymore since I upgraded TF. I tried to remember the changes, but could not find the reason. Is 'btintptr_t' defined somewhere for Linux? I could not find any reference in the code where it would be defined in the linux case...

iSLC, 2017-03-19:
If your compiler does not define `_INTPTR_T_DEFINED` and neither declares `intptr_t` somewhere (_c++11 not enabled?_) then declaring  `btintptr_t`  as `intptr_t` is equal to `btintptr_t` not being declared. 

I suppose that macro condition can be written as:
```
#if defined(INTPTR_MIN) || defined(INTPTR_MAX)
	typedef intptr_t btintptr_t;
#elif defined(_WIN64)
	typedef __int64 btintptr_t;
#else
	typedef long btintptr_t;
#endif
```

nbelakovski, 2018-02-13:
I just had this same issue compiling makesDNA on mac:
/Users/nickolai.belakovski/develop/bullet3/Extras/Serialize/makesdna/makesdna.cpp:1074:9: error: use of undeclared identifier 'btintptr_t'
It looks like the #defines that have been referenced in this thread resolve to #include <inttypes.h> for Mac. Adding `typedef long btintptr_t` fixed the compile issue, not sure if that's the desired fix though.

erwincoumans, 2018-02-13:
Why/how would you compile makesdna.cpp? The file is optional and shouldn't be compiled by default. Did you use cmake and set INTERNAL_UPDATE_SERIALIZATION_STRUCTURES to ON (it is OFF) by default?

nbelakovski, 2018-02-13:
I'm trying to extend the serialization structures, so, yea, I am compiling this on purpose.

closed 2021-04-26 by erwincoumans