bulletphysics/bullet3 #819: Almost built but failed with "cannot find -lBussIK"

jiapei100, 2016-10-07:
Hi, all:

> /usr/bin/ld: cannot find -lBussIK
> collect2: error: ld returned 1 exit status
> test/SharedMemory/CMakeFiles/Test_PhysicsClientServer.dir/build.make:1042: recipe for target 'test/SharedMemory/Test_PhysicsClientServer' failed
> make[2]: *** [test/SharedMemory/Test_PhysicsClientServer] Error 1
> make[2]: Leaving directory '....../bullet3/build'
> CMakeFiles/Makefile2:1394: recipe for target 'test/SharedMemory/CMakeFiles/Test_PhysicsClientServer.dir/all' failed
> make[1]: *** [test/SharedMemory/CMakeFiles/Test_PhysicsClientServer.dir/all] Error 2
> make[1]: Leaving directory '....../bullet3/build'
> Makefile:141: recipe for target 'all' failed
> make: *** [all] Error 2

Any suggestions?

Cheers
Pei

erwincoumans, 2016-10-07:
Can you provide the exact steps to reproduce, including platform (Windows, Linux, Mac, other), compiler, and how to invoke cmake. Are you using default settings for cmake? I can't reproduce the issue on Windows, Linux or Mac it works just fine.

erwincoumans, 2016-10-07:
Please re-open with a proper report, thanks!

M0lion, 2016-12-16:
I'm getting (I think?) the same issue, I'm on windows 10, step by step:

1. git clone https://github.com/bulletphysics/bullet3.git
2. open cmake-gui
	2.1 set src to bullet3 dir
	2.2 set build to bullet3/msvc/2015
	2.3 configure
	2.4 generator: VS 14 2015 win64 (default native compilers)
	2.5 uncheck BUILD_BULLET2_DEMOS and BUILD_CPU_DEMOS
	2.6 check BUILD_CPU_DEMOS
	2.7 configure
	2.8 generate
3. open BULLET_PHYSICS.sln
4. set configuration to MinSizeRel(doesn't seem to matter, same for release and debug)
5. build->build solution

last few lines of VS output:

26>  Generating Code...
26>LINK : fatal error LNK1181: cannot open input file 'BussIK.lib'
28>------ Build started: Project: ALL_BUILD, Configuration: MinSizeRel x64 ------
28>  Building Custom Rule C:/Programming/c#/bullet3/CMakeLists.txt
28>  CMake does not need to re-run because C:\Programming\c#\bullet3\msvc\2015\CMakeFiles\generate.stamp is up-to-date.
29>------ Skipped Build: Project: INSTALL, Configuration: MinSizeRel x64 ------
29>Project not selected to build for this solution configuration 
========== Build: 26 succeeded, 1 failed, 0 up-to-date, 2 skipped ==========

It works with default cmake settings

erwincoumans, 2016-12-16:
Then use the default cmake settings :)
2.5 check BUILD_CPU_DEMOS undoes 2.4 uncheck BUILD_CPU_DEMOS?

M0lion, 2016-12-16:
meant check USE_MSVC_RUNTIME_LIBRARY_DLL

M0lion, 2016-12-16:
I seem to have fixed it. In the lib folder there was indeed no BussIK.lib, but there was a BussIK_MinsizeRel.lib, so I changed the dependency in Test_PhysicsClientServer from BussIK.lib to BussIK_MinsizeRel.lib. Still couldn't find it, but changing it to ..\..\lib\MinSizeRel\BussIK_MinsizeRel.lib fixed it.

closed 2021-04-26 by erwincoumans