bulletphysics/bullet3 #828: Bullet Physics Vehicle gets skidding(sideways) while it's being at rest

macxfadz, 2016-10-11:
I have created a vehicle as Bullet ray-cast vehicle and simple terrain as Bullet triangle-mesh type. When my vehicle is at rest (i.e. speed < 1.0 m/s ), it's skidding sideways even if the terrain (a simple plane created using triangle-mesh or static plane) has slight inclination (slope).

I tried to figure it out whether this was caused due to friction parameters but it is not. I configured everything related even skid_info,and slip_friction etc...

Please help me to resolve this issue, in case I have done something peculiar.

For your convenience please watch this video :- https://vid.me/vJG5

I have posted this question on Bullet forum.Please follow up that link too. Thank you. Bullet forum link :- http://www.bulletphysics.org/Bullet/phpBB3/viewtopic.php?f=9&t=11373

Ming36, 2018-11-08:
I have the similar problem about the raycastvehicle. We can discuss it together. The computing method  of vehicle side impulse may be not correct.

macxfadz, 2018-11-08:
@Ming36  do you have a solution ?

Timofffee, 2019-05-11:
@macxfadz Have you solved this problem?

macxfadz, 2019-05-11:
@Timofffee sorry, I couldn't get into it.

closed 2021-04-26 by erwincoumans