bulletphysics/bullet3 #830: Install pybullet library

scpeters, 2016-10-11:
I saw the announcement about pybullet in the 2.84 prerelease, and I went to tinker with it, but I noticed that it doesn't install `libpybullet.so`. It would be nice to install it like the other libraries I think. I'm not sure how other packages would find it though.

erwincoumans, 2016-10-11:
Yes, at the moment there is no automatic way of "installing" the pybullet.so extension. In addition, it would also be nice/important to install the Example Browser. I'm not familiar with Linux/Mac release processes though.

scpeters, 2016-10-11:
For the example browser and demos, you could install them to `<prefix>/bin` if the names are generic enough, or to `<prefix>/libexec/bullet/` if you don't mind that the demos aren't in the `PATH` by default. Note that simbody puts their visualizer binary in `libexec/simbody`.

erwincoumans, 2016-10-13:
I hope someone steps up and help update the cmake build files, perhaps a pull request or patch. Until then, using pybullet and the ExampleBrowser will be a manual step, instead of easy installation.

scpeters, 2016-10-13:
Fair enough. I'm not sure when I'll have a chance to get to it, but if you leave it open maybe someone will take a crack at it.

alabd14313, 2016-10-16:
I'm in arch linux. Why I cant load pybullet from python:
`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named 'pybullet'`

Is this related to: https://github.com/bulletphysics/bullet3/issues/836

erwincoumans, 2016-10-16:
There are a few different ways of installing a python extension, usually there is a setup.py (which bypasses cmake). Cmake needs to compile the extension without prefix and using .so postfix (not dylib on Mac OSX). This is not related to #836  @alabd14313 .
See also https://m.reddit.com/r/Python/comments/26fw6y/building_a_cmake_c_application_with_python/
http://stackoverflow.com/questions/18360245/python-c-compiling-shared-library-with-cmake-and-installing-with-distutils

I may get to this sooner or later.

oroulet, 2016-11-29:
I am trying to test the pybullet stuff, I enabled it in cmake file. and it created the files libpybullet.so and libpybullet.so.2.85 but I cannot import them directly. What are we supposed to do? I cannot figure out and I am not exactly a begynner with python modules....

python3
Python 3.5.2+ (default, Sep 22 2016, 12:18:14) 
[GCC 6.2.0 20160927] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import libpybullet
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: dynamic module does not define module export function (PyInit_libpybullet)

benelot, 2016-11-29:
> File "", line 1, in

I started a description on how to install the pybullet stuff, maybe you can
find here what you are missing:
http://bulletphysics.org/mediawiki-1.5.8/index.php/Using_pybullet

oroulet, 2016-11-29:
the 'ln -s' command should really not be necessary, cmake can create a pybullet.so file by itself. I remember doing it in another c++ program.
But my dunamic module import error seems to be something else...

oroulet, 2016-11-29:
OK I found out that the link is necesary to be able to run it...

erwincoumans, 2016-11-29:
On linux and mac, the name needs to be pybullet.so
on windows pybullet.pyd.

then, make sure python can find it: in current working directory, or export PYTHONPATH =path/of/pybullet (not filename, but directory)

let us know if that helps. 

we should fix the cmake build script to do all this automatically.

erwincoumans, 2016-11-29:
so don't add the 'lib' or other prefix, the name needs to ge exactly: pybullet.so

oroulet, 2016-11-29:
Hva med å legge til : 

    SET_TARGET_PROPERTIES(pybullet PROPERTIES PREFIX "")

marcj, 2017-01-08:
I get `Fatal Python error: PyThreadState_Get: no current thread` on my OSX 10.11.3 machine. Anyone an idea?

```bash
cmake .. -DINSTALL_EXTRA_LIBS=ON -DBUILD_UNIT_TESTS=OFF -DBUILD_SHARED_LIBS=ON \
    -DBUILD_BULLET2_DEMOS=ON \
    -DBUILD_PYBULLET=ON -DBUILD_PYBULLET_NUMPY=ON -DUSE_DOUBLE_PRECISION=ON -DCMAKE_BUILD_TYPE=Release \
    -DPYTHON_EXECUTABLE:FILEPATH=/usr/local/bin/python \
    -DPYTHON_INCLUDE_DIR:PATH=/usr/include/python2.7 \
    -DPYTHON_LIBRARY:FILEPATH=/usr/lib/libpython2.7.dylib
```

```bash
marc@osx ~/bude/bullet3/build3 (master*) $ cd ../build3/examples/pybullet
marc@osx ~/bude/bullet3/build3/examples/pybullet (master*) $ ln -s libpybullet.dylib pybullet.so
marc@osx ~/bude/bullet3/build3/examples/pybullet (master*) $ python
Python 2.7.12 (default, Jun 29 2016, 14:05:02)
[GCC 4.2.1 Compatible Apple LLVM 7.3.0 (clang-703.0.31)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import pybullet
Fatal Python error: PyThreadState_Get: no current thread
[1]    29574 abort      python
```

erwincoumans, 2017-01-09:
Likely some conflict with homebrew or other python? 
What happens if you remove all the -DPYTHON_* from the cmake step?
This seems a common Mac+Python+homebrew issue, please google it and figure out a fix and report it here.

fabienbaradel, 2017-02-07:
@marcj did you fix your problem? I am encountering the same issue.

fabienbaradel, 2017-02-07:
By running the following command it works for me:
git clone https://github.com/bulletphysics/bullet3.git
cd bullet3
rm CMakeCache.txt
mkdir build_cmake
cd build_cmake
cmake -DBUILD_PYBULLET=ON -DBUILD_PYBULLET_NUMPY=OFF -DUSE_DOUBLE_PRECISION=ON -DCMAKE_BUILD_TYPE=Release -DPYTHON_EXECUTABLE:FILEPATH=/Users/fabien/anaconda/bin/python2.7 -DPYTHON_INCLUDE_DIR:PATH=/Users/fabien/anaconda/include/python2.7 -DPYTHON_LIBRARY:FILEPATH=/Users/fabien/anaconda/lib/libpython2.7.dylib ..
make
cd examples
cd pybullet
ln -s pybullet.dylib pybullet.so
python -c "import pybullet"

nimaasg, 2017-02-10:
@benelot thanks for the how-to description. 
I followed all the instructions multiple times for installing pybullet on Windows 10. The build succeeds with zero errors, but in the end there's no pybullet.pyd under cm\examples\pybullet\. Instead I can only find a pybullet_d.pyd under cm\lib\Debug\ and it doesn't work when trying to import in python. 

Any idea what might be the issue?

erwincoumans, 2017-02-10:
@nimaasg  You need to select 'Release' build.

nimaasg, 2017-02-11:
@erwincoumans I tried with -DCMAKE_BUILD_TYPE=Release option and I have CMAKE_BUILD_TYPE:STRING=Release in the CMakeCache.txt. Is there a better way to "select Release build"?

erwincoumans, 2017-02-11:
[image]

erwincoumans, 2017-02-11:
'Release' image also added in the [pybullet quickstart guide](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA) .

BluePrintRandom, 2017-02-11:
Will pybullet be able to be downloaded using PIP soon?

Pip install  pybullet?

nimaasg, 2017-03-03:
Thanks @erwincoumans. Just wanted to confirm that the Release option fixed the issue.

erwincoumans, 2017-04-03:
See https://pypi.python.org/pypi/pybullet
I created a setup.py file, so you only need to run 'pip install pybullet'
See also https://github.com/bulletphysics/bullet3/pull/1054
Only tested on Mac OSX, will fix Linux/Windows soon.

erwincoumans, 2017-04-11:
ok, pip install pybullet works on all platforms. When using Anaconda on Mac OSX, you can use:

CFLAGS="-DB3_NO_PYTHON_FRAMEWORK" pip install pybullet

closed 2017-04-11 by erwincoumans

BolunDai0216, 2018-10-09:
/usr/local/Cellar/gcc/8.2.0/include/c++/8.2.0/cwchar:44:10: fatal error: wchar.h: No such file or directory
     #include <wchar.h>
              ^~~~~~~~~
    compilation terminated.
    
    ----------------------------------------
Command "/anaconda3/bin/python -u -c "import setuptools, tokenize;__file__='/private/var/folders/26/3vt_y1v15td9p0rlbynzv72m0000gn/T/pip-install-7w79oxwl/pybullet/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /private/var/folders/26/3vt_y1v15td9p0rlbynzv72m0000gn/T/pip-record-eogpgo_m/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/var/folders/26/3vt_y1v15td9p0rlbynzv72m0000gn/T/pip-install-7w79oxwl/pybullet/

BolunDai0216, 2018-10-09:
This is the error I encountered, do you have ant idea what the issue is?

bhanu2710, 2021-02-03:
I have pybullet installed but it is still showing me error