bulletphysics/bullet3 #836: Example browser error

alabd14313, 2016-10-16:
Hello!
In arch linux, I have this error when I run bullet3_examplebrowser. It is it's packagebuild:
`# $Id$
# Maintainer: Sven-Hendrik Haase <sh@lutzhaase.com>
# Contributor: Christoph Zeiler <archNOSPAM_at_moonblade.dot.org>

pkgbase=bullet
pkgname=('bullet' 'bullet-docs')
pkgver=2.85.1
pkgrel=1
pkgdesc="A 3D Collision Detection and Rigid Body Dynamics Library for games and animation"
arch=('i686' 'x86_64')
url="http://www.bulletphysics.com/Bullet/"
license=('custom:zlib')
makedepends=('cmake' 'doxygen' 'graphviz' 'ttf-dejavu' 'mesa' 'glu' 'python' 'python-numpy')
source=("https://github.com/bulletphysics/bullet3/archive/${pkgver}.tar.gz")
md5sums=('194cf1ef1eeb7fa9ac7dc580219d3c8b')

build() {
  cd bullet3-${pkgver}

  [[ -d build ]] && rm -rf build
  mkdir build && cd build

  cmake .. \
       -DCMAKE_INSTALL_PREFIX=/usr \
       -DBUILD_SHARED_LIBS=1 \
       -DINSTALL_LIBS=1 \
       -DINSTALL_EXTRA_LIBS=1 \
       -DBUILD_PYBULLET=ON \
       -DBUILD_PYBULLET_NUMPY=ON \
       -DBUILD_OPENGL3_DEMOS=ON \
       -DCMAKE_BUILD_TYPE=Release

  make

  cd ..
  #sed -i 's/GENERATE_HTMLHELP.*//g' Doxyfile
  doxygen
}

package_bullet() {
  optdepends=('glu: for the example browser'
              'python: python bindings'
              'python-numpy: python bindings'
              'bullet-docs: documentation')

  cd bullet3-${pkgver}/build

  make DESTDIR=${pkgdir} install
  install -Dm755 examples/pybullet/libpybullet.so.2.85 ${pkgdir}/usr/lib/libpybullet.so.2.85

  install -Dm755 examples/ExampleBrowser/App_ExampleBrowser ${pkgdir}/usr/bin/bullet3_examplebrowser

  # install license
  install -Dm644 ../LICENSE.txt ${pkgdir}/usr/share/licenses/${pkgbase}/LICENSE
}

package_bullet-docs() {
  pkgdesc="Documentation for bullet"
  depends=()

  cd bullet3-${pkgver}

  # install docs
  install -Dm644 docs/GPU_rigidbody_using_OpenCL.pdf ${pkgdir}/usr/share/doc/bullet/GPU_rigidbody_using_OpenCL.pdf
  install -Dm644 docs/Bullet_User_Manual.pdf ${pkgdir}/usr/share/doc/bullet/Bullet_User_Manual.pdf
  install -Dm644 docs/BulletQuickstart.pdf ${pkgdir}/usr/share/doc/bullet/BulletQuickstart.pdf
  cp -r html ${pkgdir}/usr/share/doc/bullet/html
}
# vim: sw=2 ts=2 et:
`

Do you have any suggestion?

erwincoumans, 2016-10-16:
The BulletExampleBrowser depends on libBulletExampleBrowserLib, so if you install it system-wide, you need to 'install' that library too, including various other libraries.
If you use the original Bullet premake or cmake build systems, it should all work fine.

erwincoumans, 2016-10-16:
Could one also build/run that packagebuild install (PKGBUILD.txt) on Ubuntu, if so how?

alabd14313, 2016-10-16:
@erwincoumans , here is some help to create a PPA in ubuntu, though is not very straightforward:
https://wiki.debian.org/HowToPackageForDebian
http://askubuntu.com/questions/71510/how-do-i-create-a-ppa

closed 2021-04-26 by erwincoumans