bulletphysics/bullet3 #843: Compound physics Armature Actors

BluePrintRandom, 2016-10-28:
So I have found an easy way to get an accurate physics shape for actors in blender

I have a compound root object - the armature to be animated parents to this,
each bone that is to have physics has a 'bone tag' parented to it, and a shild shape
that approximates the shape of the bone

every frame I remove parent on all the child shapes, move them to where the bone tag is, and match it's rotation,

this allows my actors to go up and down stairs by applying a forward servo motion while playing a stair walk animation.

this system works great, but when a child shape is contacting, and is moved, produces no friction,

I purpose a build in system that does all this under the hood but also computes the friction during the move maybe using substeps?

here is the system in action - https://www.youtube.com/watch?v=cx_9F0RNtzc
however she will not walk forward only by playing an animation - I have to combine the animation with a forward sero motion.

benelot, 2016-10-28:
What you describe totally sounds like an issue dicussed only on the bullet physics forum and not the github developer's page. Give it a try there, people will help you. And do not forget to close the issue here.

BluePrintRandom, 2016-10-28:
Moving to forum**

benelot, 2016-10-30:
Oh, I am sorry, I did not see that you proposed a new built-in system. I was somehow carried away by the rest. Feel free to open your issue again.

BluePrintRandom, 2016-10-31:
basically we need a method that moves and rotates the child shapes each frame, and computes the forces this would generate based on contacts during the substeps I think*

has anyone here ever delt with this?
(animating an actors physics shape using compound child shapes?)

closed 2021-04-26 by erwincoumans