bulletphysics/bullet3 #846: issue with btMatrix3x3::diagonalized()

erwincoumans, 2016-10-30:
See also http://bulletphysics.org/Bullet/phpBB3/viewtopic.php?f=9&t=11407
more details will follow

erwincoumans, 2016-10-30:
Here is a snippet to reproduce the issue:

	btMatrix3x3 mat(6.66666674e-05, 9.09494702e-13, 0.000000000,
	-9.09494702e-13, 6.66666674e-05, 0.000000000,
	0.000000000, 0.000000000, 6.66666674e-05);

	btScalar threshold = 1e-4;// 130;
	int maxStep = 120;
	btMatrix3x3 rot;
	mat.diagonalize(rot, threshold, maxStep);

scpeters, 2016-10-31:
the header file says that a [symmetric matrix is assumed](https://github.com/bulletphysics/bullet3/blob/3d813b2f1bb7c94346aae2d2b650f82ac96f2085/src/LinearMath/btMatrix3x3.h#L658), but the example matrix is not symmetric

scpeters, 2016-10-31:
maybe it shouldn't matter if the asymmetry is small?

hyyou, 2016-12-07:
scpeters, it is a symmetric matrix.

hyyou, 2016-12-07:
You are right, scpeters.  I (the person who reported this bug), didn't read that comment.  I can now rest in peace, thank.

erwincoumans, 2017-02-26:
The newer version of diagonalize is more robust and should fix your issue. See https://github.com/bulletphysics/bullet3/commit/0131754173f38d63659c9d8c57ce2d1fb03cf51c

closed 2017-02-26 by erwincoumans