bulletphysics/bullet3 #869: Fix memory leaks in ExampleBrowser, pybullet and SharedMemoryAPI + add unit test

erwincoumans, 2016-11-21:
There are some memory leaks in the example code (Example Browser, pybullet, Shared Memory API). This needs to be fixed, and Continuous Integration tests for memory leaks added to Travis / AppVeyor.

DavidDeAngelo, 2017-01-08:
I noticed the GUIHelperInterface has a couple methods that create objects, but it lacks a corresponding remove function, for example createPhysicsDebugDrawer. If you create multiple CommonExampleInterfaces and delete them, you'll have some left over objects. While its doubt its likely that someone would do that, I can't think of any good reason to have create functions without remove functions.

erwincoumans, 2017-01-19:
The one who allocates stuff is responsible to delete it properly, so creating multiple copies etc. is considered a user error.

erwincoumans, 2017-01-19:
The leaks in Example browser etc should be fixed, reopen new issue if there are new leaks found.

erwincoumans, 2020-08-25:
The leaks should be fixed a while ago, closing.

closed 2020-08-25 by erwincoumans