bulletphysics/bullet3 #871: btOverlapFilterCallback::needBroadphaseCollision invocations and proper usage

SeanCurtis-TRI, 2016-11-23:
I had naively assumed that the `btOverlapFilterCallback::needBroadphaseCollision` method would be invoked repeatedly during calls to `btCollisionWorld::performDiscreteCollisionDetection`.  However, this appears to *not* be the case. It seems it contributes to populating the cache, but then the collision detection relies wholly on the cache being correct.

Specifically, in our usage, we're using a `btDbvhBroadphase` class as the collision world's implementation of the broadphase interface.  The implementation is a big difficult to follow, but it seems that it only uses the `btOverlapFilterCallback` during restructuring the hierarchy (e.g., when adding collision objects).  That means, if the filter callback's functionality is dependent on program state that is *not* in place when collision objects are added to the collision world, it will return the wrong results and pairs of collision objects which, are intended to be filtered out, would be included.  And only as the bodies move and the hierarchy has to update and re-evaluate aspects of the pair cache will the overlapping pair filter get re-evaluated for a given pair.

This type of behavior is not clear or documented.  Can you elaborate on the requirements for properly using this overlapping filter callback?

erwincoumans, 2016-11-23:
Indeed, this is expected behaviour: broadphase pairs are only add/removed incrementally. 

External changes to the collision objects/shapes (aside from setting the world transform) are invisible. You have a few options when you rely on such external changes: 
1) remove the object from the world, make the change, re-insert to the world.
2) 'notify' the broadphase explicitly
operation.
		m_dynamicsWorld->getBroadphase()->getOverlappingPairCache()->cleanProxyFromPairs(staticBody->getBroadphaseHandle(),getDynamicsWorld()->getDispatcher());

SeanCurtis-TRI, 2016-11-23:
The third option would be making sure the registered overlap filter callback has all the data it needs *before* adding collision objects to the world.

Of those three options, which would you recommend?  Particularly in the light of the fact that we would have to "reset" an arbitrarily large number of collision objects?

I don't know what the meaningful costs of your two suggestions are.  The cost of the third option is largely one of one-time programming logistics and want to get a sense of what the "best" option would be.  

(p.s. thanks for the *fast* response.)

erwincoumans, 2016-11-23:
Option 3 doesn't make much sense to me, since I expected the overlap filter callback to have all the data it needs when it is called, otherwise how can it report a meaningful result?

What kind of changes to you plan to make, while the object is inside the world?

SeanCurtis-TRI, 2016-11-23:
Our problem is purely one of logistics. 

We define our world by reading input files (urdf/sdf).  As we read specifications for collision geometry we push it into the collision world (very much an on-the-fly) kind of thing.  After we're done reading the file, we do some post processing to evaluate what we've read in and classify which objects should actually be filtered out.  By the end of this post-processing, the object filter now has enough smarts to recognize when a pair should be filtered out.  But the damage has already been done. 

So, in our case, it's not that we're particularly changing the collision objects.  The problem is that the code was written without an understanding of Bullet's internal workings.  The thinking was that as long as the filter was in place *before* collision queries were performed it was good.  This was a mistake.

So, what we have to do is accumulate all of the collision geometry, analyze it, and then, finally, add it to the collision world now that the overlap filter has sufficient data to provide the desired effect.

Make sense?

SeanCurtis-TRI, 2016-12-12:
Issue is resolved based on discussion; clarification was sufficient.

closed 2016-12-12 by SeanCurtis-TRI