bulletphysics/bullet3 #878: bug in btMultiBody::clearVelocities()

12michi34, 2016-12-01:
btMultiBody::clearVelocities() does currently 

void btMultiBody::clearVelocities()
{
	for (int i = 0; i < 6 + getNumLinks(); ++i) 
	{
		m_realBuf[i] = 0.f;
	}
}
index goes from 0 to 6 + getNumLinks().
But m_realBuf gets allocated as 
m_realBuf.resize(6 + m_dofCount + m_dofCount*m_dofCount + 6 + m_dofCount);	

No consider a situation where getNumLinks()> m_dofCount ... Then memory gets trampled.

Simple repro is a btMultiBody with only fixed constrtaints (=m_dofCount =0, and getNumLinks() > 0).

Cheers,

Michi

erwincoumans, 2016-12-11:
thanks for the report, that looks like a bug indeed, will look into fixing it asap.

erwincoumans, 2016-12-11:
Fixed in https://github.com/bulletphysics/bullet3/pull/886/commits/710ac09b565b7c5759c09456b3861b42ed833e24

closed 2016-12-11 by erwincoumans