bulletphysics/bullet3 #879: bus error with example browser and pybullet

oroulet, 2016-12-01:
I get a lot of random crash at startup with applications using the UI:
./App_ExampleBrowser 
X11 functions dynamically loaded using dlopen/dlsym OK!
glewXInit dynamically loaded using dlopen/dlsym OK
glewXInit OK
Creating context
Created GL 3.0 context
Direct GLX rendering context obtained
Making context current
glewOpenGL11Init dynamically loaded using dlopen/dlsym OK
glewOpenGL11Init OK!
GL_VENDOR=Intel Open Source Technology Center
GL_RENDERER=Mesa DRI Intel(R) HD Graphics 5500 (Broadwell GT2) 
GL_VERSION=4.3 (Core Profile) Mesa 12.0.3
GL_SHADING_LANGUAGE_VERSION=4.30
pthread_getconcurrency()=0
b3Printf: Selected demo: Convex stack
Bus error

or from pybullet:
import pybullet as p
p.connect(p.GUI)   # OK
p.loadURDF("plane.urdf")  #crash!

Today it is more than random, I cannot start any application.
Does anyone else get that kind of errors?

under gdb

Program received signal SIGBUS, Bus error.
__memmove_avx_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:448
448	../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S: No such file or directory.
(gdb) where
#0  __memmove_avx_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:448
#1  0x00007ffff229a334 in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
#2  0x00007ffff7251bc4 in GLInstancingRenderer::registerShape(float const*, int, int const*, int, int, int) ()
   from /home/olivier/dev/bullet3/build/examples/OpenGLWindow/libOpenGLWindow.so
#3  0x00007ffff7ba90b0 in OpenGLGuiHelper::createCollisionShapeGraphicsObject(btCollisionShape*) ()
   from /home/olivier/dev/bullet3/build/examples/ExampleBrowser/libBulletExampleBrowserLib.so.2.85
#4  0x00007ffff7baac7f in OpenGLGuiHelper::autogenerateGraphicsObjects(btDiscreteDynamicsWorld*) ()
   from /home/olivier/dev/bullet3/build/examples/ExampleBrowser/libBulletExampleBrowserLib.so.2.85
#5  0x0000555555661fbe in BenchmarkDemo::initPhysics() ()
#6  0x00007ffff7ba06d2 in selectDemo(int) ()
   from /home/olivier/dev/bullet3/build/examples/ExampleBrowser/libBulletExampleBrowserLib.so.2.85
#7  0x00007ffff7ba3ae9 in OpenGLExampleBrowser::init(int, char**) ()
   from /home/olivier/dev/bullet3/build/examples/ExampleBrowser/libBulletExampleBrowserLib.so.2.85
#8  0x00005555555a8ff6 in main ()

erwincoumans, 2016-12-01:
Those are most likely graphics driver issues. Do you have another machine with different gpu/drivers? Also, make sure to do a clean build, no custom settings.

oroulet, 2016-12-01:
Ok. Thank you. I wanted to hear if this was something others have, or
something specific on my pc

erwincoumans, 2016-12-01:
Can you provide a few FULL stack traces of the crash using gdb?

Try running the Bullet ExampleBrowser using the --opengl2 option and see if you still get crashes.

oroulet, 2016-12-01:
I will, next time it crashes, now it is working again...

erwincoumans, 2016-12-01:
feel free to add further info to this thread, closing it for now.

closed 2016-12-01 by erwincoumans

karaoulanis, 2017-11-29:
For the following lines based on ``hello_pybullet.py ``
```python
import pybullet as p
from time import sleep

physicsClient = p.connect(p.GUI)
```
I get the following bus error:
```$ python3 hello_pybullet.py 
pybullet build time: Nov 29 2017 09:19:55
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=4
argv[0] = unused
argv[1] = 
argv[2] = --logtostderr
argv[3] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
glewXInit dynamically loaded using dlopen/dlsym OK
glewXInit OK
Creating context
Created GL 3.0 context
Direct GLX rendering context obtained
Making context current
glewOpenGL11Init dynamically loaded using dlopen/dlsym OK
glewOpenGL11Init OK!
GL_VENDOR=Intel Open Source Technology Center
GL_RENDERER=Mesa DRI Intel(R) Ivybridge Desktop 
GL_VERSION=3.3 (Core Profile) Mesa 11.2.0
GL_SHADING_LANGUAGE_VERSION=3.30
pthread_getconcurrency()=0
Bus error
```
If I enable ``opengl2``, as in:
```python
import pybullet as p
from time import sleep

physicsClient = p.connect(p.GUI, options="--opengl2")
```
everything seems to be working fine:
```
 $ python3 hello_pybullet.py 
pybullet build time: Nov 29 2017 09:19:55
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=4
argv[0] = unused
argv[1] = --opengl2
argv[2] = --logtostderr
argv[3] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
glewXInit dynamically loaded using dlopen/dlsym OK
glewXInit OK
4

	visual 0x20 selected
glewOpenGL11Init dynamically loaded using dlopen/dlsym OK
glewOpenGL11Init OK!
GL_VENDOR=Intel Open Source Technology Center
GL_RENDERER=Mesa DRI Intel(R) Ivybridge Desktop 
GL_VERSION=3.0 Mesa 11.2.0
GL_SHADING_LANGUAGE_VERSION=1.30
pthread_getconcurrency()=0
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
numActiveThreads = 0
stopping threads
stopThreads: Thread 0 used: 1
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
stopThreads: Thread 0 used: 1
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed
```
Let me know if you need any additional info.