bulletphysics/bullet3 #883: build failed with premake on App_TinyRenderer

consultit, 2016-12-04:
I tried to build using premake, issuing these commands:

`cd build3`
`./premake4_linux64 gmake`
`cd gmake`
`make config=release64`

and I got these errors:

`
...
==== Building App_TinyRenderer (release64) ====
main.cpp
../../examples/TinyRenderer/main.cpp: In function ‘int main(int, char**)’:
../../examples/TinyRenderer/main.cpp:150:61: error: no match for ‘operator=’ (operand types are ‘b3AlignedObjectArray<float>’ and ‘float’)
                 renderData.m_shadowBuffer[x+y*textureWidth] = -1e30f;
                                                             ^
../../examples/TinyRenderer/main.cpp:150:61: note: candidate is:
In file included from ../../examples/OpenGLWindow/../OpenGLWindow/GLInstancingRenderer.h:19:0,
                 from ../../examples/OpenGLWindow/SimpleOpenGL3App.h:4,
                 from ../../examples/TinyRenderer/main.cpp:1:
../../src/Bullet3Common/b3AlignedObjectArray.h:59:43: note: b3AlignedObjectArray<T>& b3AlignedObjectArray<T>::operator=(const b3AlignedObjectArray<T>&) [with T = float]
  B3_FORCE_INLINE b3AlignedObjectArray<T>& operator=(const b3AlignedObjectArray<T> &other)
                                           ^
../../src/Bullet3Common/b3AlignedObjectArray.h:59:43: note:   no known conversion for argument 1 from ‘float’ to ‘const b3AlignedObjectArray<float>&’
App_TinyRenderer.make:163: recipe for target 'obj/x64/Release/App_TinyRenderer/main.o' failed
make[1]: *** [obj/x64/Release/App_TinyRenderer/main.o] Error 1
Makefile:36: recipe for target 'App_TinyRenderer' failed
make: *** [App_TinyRenderer] Error 2
`

consultit, 2016-12-11:
Found the error on `examples/TinyRenderer/main.cpp`.
Attached is a patch (computed with: `git diff examples/TinyRenderer/main.cpp`).
Unzip and apply with ```git apply  TinyRenderer_main_cpp.patch```

[TinyRenderer_main_cpp.patch.zip](https://github.com/bulletphysics/bullet3/files/644334/TinyRenderer_main_cpp.patch.zip)

erwincoumans, 2016-12-11:
should be fixed by now, thanks for the report!

closed 2016-12-11 by erwincoumans