bulletphysics/bullet3 #896: GJK/EP overly sensitive to rotations

SeanCurtis-TRI, 2016-12-20:
Issue #889 provides example code of attempting to find the nearest point on a `btCollisionObject` to a point.  The point is modeled as a zero-radius sphere.  It illustrates that as the point moves closer to the `btCollisionObject` the normal direction reverses direction.

The sphere is *mistakenly* initialized with a quaternion of <1, 0, 0, 0> instead of the more meaningful "identity" quaternion <0, 0, 0, 1>.  In fact, correcting the quaternion resolves the observed behavior.

The bug is that because the query point is modeled as a sphere, the answer shouldn't change.  Spheres are rotation invariant; but bullet does not exhibit this trait.

erwincoumans, 2016-12-20:
This should not happen in the latest trunk version of Bullet.
Did you try using the very latest github version, or are you using an older version? ( in particular this commit https://github.com/bulletphysics/bullet3/commit/e0c46f4b1a776842e61da9f8cc9cd61855225840)

Also, don't use //  convexConvex.setIgnoreMargin(true);

erwincoumans, 2018-12-23:
I think this should be fixed by now. If not, please re-open with a clear repro case. Where was this <1, 0, 0, 0> quaternion exactly?
Thanks for the reports!

closed 2018-12-23 by erwincoumans