bulletphysics/bullet3 #897: idea: floating point operation count by a custom floating point class in btScalar.h

kkimdev, 2016-12-24:
We can create a floating point number class that counts operations, and plug in to btScalar to see how many operations are performed exactly.  I'm considering creating such class.

kkimdev, 2016-12-24:
@erwincoumans do you have any opinion on this??

BluePrintRandom, 2016-12-25:
it's chistmass :)

maybe try back later

erwincoumans, 2017-01-07:
Yes, there are some interesting possibilities using a custom floating point type: using a fixed-point type for cross-platform determinism (identical results), using auto-diff types for automatic differentiation (machine learning) etc. I am considering if Bullet 2.x, 3.x (OpenCL) or future 4.x API is the best for this. If you manager to get something working, it may be a huge patch that is too tricky to review. Nevertheless, it would be interesting and possibly useful.

kkimdev, 2017-01-08:
I was able to replace btScalar & b3Scalar with my automatic differentiation library.  And the below is the output from the tutorial http://www.bulletphysics.org/mediawiki-1.5.8/index.php/Hello_World .  The second column is the sphere height differentiated by the gravity constant (-10).

It was mostly straightforward change but there were also many parts that I wasn't sure how to fix, so I just commented out those.  You can see my bullet3 fork for the source code, but I should first clean up a bit. Then I will ask some questions on the unresolved parts.

It was really exciting to see that actually working :).

```
sphere height: 50       0
sphere height: 49.9972  0.000277778
sphere height: 49.9917  0.000833333
sphere height: 49.9833  0.00166667
sphere height: 49.9722  0.00277778
sphere height: 49.9583  0.00416667
sphere height: 49.9417  0.00583333
sphere height: 49.9222  0.00777778
sphere height: 49.9     0.01
sphere height: 49.875   0.0125
sphere height: 49.8472  0.0152778
sphere height: 49.8167  0.0183333
sphere height: 49.7833  0.0216667
sphere height: 49.7472  0.0252778
sphere height: 49.7083  0.0291667
sphere height: 49.6667  0.0333333
sphere height: 49.6222  0.0377778
sphere height: 49.575   0.0425
sphere height: 49.525   0.0475
sphere height: 49.4722  0.0527778
sphere height: 49.4167  0.0583333
sphere height: 49.3583  0.0641667
sphere height: 49.2972  0.0702778
sphere height: 49.2333  0.0766667
sphere height: 49.1667  0.0833334
sphere height: 49.0972  0.0902778
sphere height: 49.025   0.0975
sphere height: 48.95    0.105
sphere height: 48.8722  0.112778
sphere height: 48.7917  0.120833
sphere height: 48.7083  0.129167
sphere height: 48.6222  0.137778
sphere height: 48.5333  0.146667
sphere height: 48.4417  0.155833
sphere height: 48.3472  0.165278
sphere height: 48.25    0.175
sphere height: 48.15    0.185
sphere height: 48.0472  0.195278
sphere height: 47.9417  0.205833
sphere height: 47.8333  0.216667
sphere height: 47.7222  0.227778
sphere height: 47.6083  0.239167
sphere height: 47.4917  0.250833
sphere height: 47.3722  0.262778
sphere height: 47.25    0.275
sphere height: 47.125   0.2875
sphere height: 46.9972  0.300278
sphere height: 46.8667  0.313333
sphere height: 46.7333  0.326667
sphere height: 46.5972  0.340278
sphere height: 46.4583  0.354167
sphere height: 46.3167  0.368333
sphere height: 46.1722  0.382778
sphere height: 46.025   0.3975
sphere height: 45.875   0.4125
sphere height: 45.7222  0.427778
sphere height: 45.5667  0.443333
sphere height: 45.4083  0.459167
sphere height: 45.2472  0.475278
sphere height: 45.0833  0.491667
sphere height: 44.9167  0.508333
sphere height: 44.7472  0.525278
sphere height: 44.575   0.5425
[... 165 lines cut ...]
sphere height: 0.999953 -nan
sphere height: 0.999963 -nan
sphere height: 0.99997  -nan
sphere height: 0.999976 -nan
sphere height: 0.999981 -nan
sphere height: 0.999985 -nan
sphere height: 0.999988 -nan
sphere height: 0.99999  -nan
sphere height: 0.999992 -nan
sphere height: 0.999994 -nan
sphere height: 0.999995 -nan
sphere height: 0.999996 -nan
sphere height: 0.999997 -nan
sphere height: 0.999997 -nan
sphere height: 0.999998 -nan
sphere height: 0.999998 -nan
sphere height: 0.999999 -nan
sphere height: 0.999999 -nan
sphere height: 0.999999 -nan
sphere height: 0.999999 -nan
sphere height: 0.999999 -nan
sphere height: 1        -nan
sphere height: 1        -nan
sphere height: 1        -nan
```

kkimdev, 2017-01-17:
These are the issues that I noticed.  Mostly minor stuff.

1. Needed to disable SSE, NEON, sqrt approximation, and OpenCL.
2. Some places had `std::*` math functions so I had to remove `std::`.  I think this is not intended and actually a bug? e.g., `const idScalar cos_gamma = std::cos(gamma);`
3. Needed to add quite of few explicit conversions, e.g., 
```C++
dof->m_angularStopCFM.m_floats[i]          = (float) m_angularLimits[i].m_stopCFM;
const double a = randomFloat(-(float) BT_ID_PI, (float) BT_ID_PI);
const idScalar alpha_DH = btScalar(5.0) * BT_ID_PI / m_num_bodies;
``` 
If the custom floating type supports implicit conversion then some are not necessary but I strongly prefer not to have implicit conversions to prevent unintended conversions.

4. A couple of places had `union` and anonymous `struct` . e.g., `b3QuadWord.h`
```C++
union
{
	b3Scalar	m_floats[4];
	struct {b3Scalar x,y,z,w;};
};
```
I didn't know this, but looks like anonymous structs can't have a member with a constructor.
`error: member ‘b3Scalar b3QuadWord::<anonymous struct>::x’ with constructor not allowed in anonymous aggregate
   struct {b3Scalar x,y,z,w;};`
and I think we shouldn't use `union` for custom floating point types due to constructor & destructor.

Overall, the issues are minor and local so it's very doable.  @erwincoumans any thoughts on how we can move this forward?

Few silly questions:
1. From what I've read, Bullet 3 API is not production ready yet?  Then is there ETA?  And can it be used without OpenCL?
2. Are Bullet2 and Bullet3 code paths independent as far as btScalar and b3Scalar are concerned?

erwincoumans, 2017-01-26:
Thanks a lot for sharing the results. My preference would be to re-implement a prototype physics engine from scratch with support for automatic differentiation and fixed-point arithmetic. . I'll get back to this.