bulletphysics/bullet3 #903: pybullet SHARED_MEMORY seems not to work anymore on Linux

benelot, 2017-01-02:
The SHARED_MEMORY version of pybullet seems not to work anymore on Linux.

If I open an example browser and set it to physics server (Shared Mem) I already get a message in the console below that it could not connect to the client (no idea why this would come up there).

If I then open a python shell and run:

     import pybullet as p
     p.connect(p.SHARED_MEMORY)

I get again this:

     b3Error[<myfolder>/bullet3-examples/examples/SharedMemory/PosixSharedMemory.cpp,60]:
     shmget errorb3Error[<myfolder>/bullet3-examples/examples/SharedMemory/PhysicsClientSharedMemory.cpp,208]:
     Cannot connect to shared memory

Can anybody confirm this?

Connecting to p.GUI still works.

erwincoumans, 2017-01-02:
Did you try a reboot, it sometimes helps issues with shared memory.

I'll check it soon (on vacation now)

benelot, 2017-01-02:
In my case the restart did not help unfortunately. Happy new year and enjoy your vacations (You definitely deserve it as some people told me at NIPS)! I will continue to work on my pybullet examples. Meet you again later!

erwincoumans, 2017-01-03:
It would be nice to first re-create the Gym cartpole example, identical to the MuJoCo one. (before going more complex)
Also, do you mind doing a bisect to see what commit breaks your shared memory on Linux? Thanks!

erwincoumans, 2017-01-04:
I briefly tested the latest github version on Linux and Mac, and it works just fine.
Running this script: build_and_run_cmake_pybullet_double.sh
then run examples/ExampleBrowser/App_ExampleBrowser, switch from BasicDemo to Experimental/Physics Server -> connection OK
Then cd examples/pybullet

```
python
import pybullet as p
p.connect(p.SHARED_MEMORY)
p.loadURDF("r2d2.urdf")

```
works all fine.
Can you do a fresh checkout and run above steps, after a reboot, not running any other apps on Linux?

erwincoumans, 2017-01-04:
In the example browser, make sure to select Experiments/Physics Server.
NOT "Physics Client (Shared Memory)"
[image]

benelot, 2017-01-04:
Oh I am sorry, I did not want to interrupt your vacation! I was about to do the bisect this evening. Oh, the latter is probably the issue. I will try it and then report back. Thanks for helping me out.

benelot, 2017-01-04:
Ok, the problem is solved. I was trying to connect to the "Physics Server (Shared Mem)" instead of the Physics Server as you describe it above. All works as expected.

closed 2017-01-04 by benelot

erwincoumans, 2017-01-04:
Good. Actually, there is no 'Physics Server (Shared Mem)", there is only a "Physics Client (Shared Memory)', I guess you used the latter.
Also, when you use premake instead of cmake, you get a few extra binaries: some standalone physics servers and a UDP networked physics servers. It would be good to also add those projects to cmake. It is easier than using the example browser for this.

erwincoumans, 2017-01-04:
Thanks!. Actually, you can use the loadMJCF to load MuJoCo xml files, so it is not so much about replacing the XML file, but implementing the bridge to OpenAI GYM to use pybullet to solve that particular CartPole environment. I will improve the MJCF importer so it works well with all GYM MJCF xml files.

benelot, 2017-01-05:
Oh right, I saw that there was such a load method. Let me see if it works with the file above. If not, I can report back in its separate future pull request.