bulletphysics/bullet3 #907: I got following error when I run python3 robotcontrol.py

spk921, 2017-01-05:
I got following error when I run python3 robotcontrol.py 

Traceback (most recent call last):
  File "robotcontrol.py", line 26, in <module>
    p.getContactPointData(husky)
AttributeError: module 'pybullet' has no attribute 'getContactPointData'

benelot, 2017-01-05:
Do you have the newest version from the repository? Depending on your OS, you can use either the bat or sh script to properly compile pybullet.

spk921, 2017-01-05:
Yes, I am using Ubuntu 16.04.

erwincoumans, 2017-01-05:
The API changed into getContactPoints, the scripts need update. See Bullet/docs/pybullet_quickstart_guide.pdf for details.

closed 2017-01-06 by erwincoumans

spk921, 2017-01-06:
Thank you for your comments. However, I encounter another error after pull and recompiled. 
Is this for my case only? 

b3Printf: rear_bumper_link
(0, b'chassis_joint', 4, -1, -1, 0, 0.0, 0.0)
(1, b'imu_joint', 4, -1, -1, 0, 0.0, 0.0)
(2, b'front_left_wheel', 0, 7, 6, 1, 0.0, 0.0)
(3, b'front_right_wheel', 0, 8, 7, 1, 0.0, 0.0)
(4, b'rear_left_wheel', 0, 9, 8, 1, 0.0, 0.0)
(5, b'rear_right_wheel', 0, 10, 9, 1, 0.0, 0.0)
(6, b'top_plate', 4, -1, -1, 0, 0.0, 0.0)
(7, b'user_rail', 4, -1, -1, 0, 0.0, 0.0)
(8, b'front_bumper', 4, -1, -1, 0, 0.0, 0.0)
(9, b'rear_bumper', 4, -1, -1, 0, 0.0, 0.0)
finished, #skip = 2823, skip1 = 0
numActiveThreads = 0
stopping threads
stopThreads: Thread 0 used: 1
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
stopThreads: Thread 0 used: 1
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed

erwincoumans, 2017-01-06:
What is the error, it seems to have finished just fine?

spk921, 2017-01-06:
Sorry my bad. It terminates when the task is done. Thank you for your comments.

erwincoumans, 2017-01-06:
Thanks for your feedback, the scripts are very simple/basic, just to test new pybullet features. The quadruped.py script has a simple quadruped hopping, please try it.
We'll create some nicer scripts for the reinforcement learning with OpenAI Gym and perhaps DeepMind Lab.

spk921, 2017-01-06:
That is really awesome. I am using MuJoCo with OpenAI gym. I was looking around physic engine that can use with binding to OpenAI gym. I found gazebo and ROS with OpenAI gym. However, in MuJoCo website bench mark http://www.mujoco.org/benchmark.html claims their platform outperform. I assumed that they used older version. When do you think it will be done? Is there are beta version to try yet?

spk921, 2017-01-06:
For the binding I used :https://github.com/openai/mujoco-py 
And combined existing xml file on MuJoCo Forum: http://www.mujoco.org/forum/index.php?resources/ 
I never tried on real robot. Just using simulation for now. 
I can send you sample results and xml file if you send me e-mail address.

PR #912 (2017-01-06): Update robotcontrol.py script
I did the change again in case you forgot to fix it directly on github.

Quick-update for the robotcontrol.py script.

This fix closes #907.
commit fa2a7348 (2017-01-06): Update robotcontrol.py script. Closes #907.