bulletphysics/bullet3 #909: Accessing the links of a URDF body

benelot, 2017-01-05:
I have an URDF file like here:

<?xml version="1.0"?>
	<robot name="cartpolev0">
	  <link name="cart">
	    <contact>
	      <lateral_friction value="1.0"/>
	      <rolling_friction value="0.0"/>
	      <inertia_scaling value="3.0"/>
	      <contact_cfm value="0.0"/>
	      <contact_erp value="1.0"/>
	    </contact>
	    <inertial>
	      <origin rpy="0 0 0" xyz="0 0 0"/>
	      <mass value="1"/>
	      <inertia ixx="1" ixy="0" ixz="0" iyy="1" iyz="0" izz="1"/>
	    </inertial>
	    <visual>
	      <origin rpy="0 0 0" xyz="0.0 0.0 0.0"/>
	      <geometry>
	        <box size="0.2 0.2 0.05"/>
	      </geometry>
	      <material name="blockmat">
	        <color rgba="0.9 0.2 0.1 1"/>
	      </material>
	    </visual>
	    <collision>
	      <origin rpy="0 0 0" xyz="0.0 0.0 0.0"/>
	      <geometry>
	        <box size="0.2 0.2 0.05"/>
	      </geometry>
	    </collision>
	  </link>

	  <joint name="cart_pole_joint" type="continuous">
		<axis xyz="0 0 1"/>
		<parent link="cart"/>
		<child link="pole"/>
		<origin rpy="1.57 0 0" xyz="0 0 0"/>
		<limit effort="100" velocity="100"/>
		<joint_properties damping="0.0" friction="0.0"/>
	  </joint>

	  <link name="pole">
	    <contact>
	      <lateral_friction value="1.0"/>
	      <rolling_friction value="0.0"/>
	      <inertia_scaling value="3.0"/>
	      <contact_cfm value="0.0"/>
	      <contact_erp value="1.0"/>
	    </contact>
	    <inertial>
	      <origin rpy="0 0 0" xyz="0 0 0"/>
	      <mass value="5"/>
	      <inertia ixx="1" ixy="0" ixz="0" iyy="1" iyz="0" izz="1"/>
	    </inertial>
	    <visual>
	      <origin rpy="0 0 0" xyz="0.0 0.25 0.0"/>
	      <geometry>
	        <box size="0.02 0.5 0.02"/>
	      </geometry>
	      <material name="blockmat">
	        <color rgba="0.2 0.7 0.1 1"/>
	      </material>
	    </visual>
	    <collision>
	      <origin rpy="0 0 0" xyz="0.0 0.25 0.0"/>
	      <geometry>
	        <box size="0.02 0.5 0.02"/>
	      </geometry>
	    </collision>
	  </link>
	 </robot>

Now after I loaded it into pybullet, I am trying to access the different links of the cartpole, namely the cart and the pole separately. Can this be done? I am looking for a similar method to getNumJoints and other for Links, also the getLinkState interestingly does return reasonable information only for index 0. So there seems to be no second link. Do I have to load the two bodies separately and then somehow link them?

erwincoumans, 2017-01-05:
Each multibody has a base (or root link), that you need to access using getBasePositionAndOrientation. Then each link is connected to its parent through a joint, so the number of joints == number of links.
For example, if you create a cube without any joints (or links), it will have a base with 0 joints. The 'link index' of the base is -1. Aside from the base, each link+joint has link indices >= 0.

Your cartpole has 1 link, connected to the base using a joint. For more info see https://github.com/bulletphysics/bullet3/blob/master/docs/pybullet_quickstartguide.pdf . We should add some pictures and examples to clarify this.

closed 2017-01-05 by erwincoumans

themightyoarfish, 2019-05-03:
But how are the link indices assigned? One link may have multiple joints attached to it, so there can be branching. Are the indices just the order in which the links appear in the urdf file?

erwincoumans, 2019-05-03:
@themightyoarfish each link is attached by exactly 1 joint to the parent, so the number of links == number of joints. The link indices can be different than the order in the URDF file. The loadURDF command has a flag (pybullet.URDF_MAINTAIN_LINK_ORDER) to maintain the order of links in the URDF, but in that case you are responsible for making sure the parent links are in the URDF file before child links.

themightyoarfish, 2019-05-28:
Is there anything more to it? When the only changes I make are passing this flag (`flags=pybullet.URDF_MAINTAIN_LINK_ORDER`) and rearranging the Joints in the urdf, the model shows briefly and then disappears. Does an STL file used for the geometry need to obey any constraints in such a case?