bulletphysics/bullet3 #922: pybullet: getLinkState segfaults in the newest version

benelot, 2017-01-19:
I am loading a simple urdf and try to get 

c =p.loadURDF('cartpole.urdf',0,0,0) # this has two rigidbodies and one joint connecting them
p.getLinkState(c,0)

but it segfaults me out of python. Am I using it wrongly? Or what is happening?

benelot, 2017-01-19:
Maybe it is a problem on my side, the quadruped.py segfaults me as well in the p.setJointMotorControl

erwincoumans, 2017-01-19:
Did you make a clean build? Does it happen in all modes (DIRECT, GUI, SHARED_MEMORY,UDP) or just one? Things work just fine here. Try making a unit test that fails (Bullet has unit tests that run in CI under the test directory), and I'll fix it.

benelot, 2017-01-19:
I made a clean build. It happened in GUI mode. I found out that the newest pybullet needs the newest gcc compiler 6. Otherwise I was getting "Importation fails due to undefined symbol: __cpu_model" when trying to import pybullet in python (This seems to be a problem with gcc 5, which is all you get by default in ubuntu/linux mint). Now everything is working fine again. Thanks for the help and sorry for the disturbance.

I updated to the newest gcc using this (in case anyone finds this):
https://gist.github.com/application2000/73fd6f4bf1be6600a2cf9f56315a2d91

closed 2017-01-19 by benelot