bulletphysics/bullet3 #923: RPATH problem on macOS 10.12

claudiofantacci, 2017-01-19:
Hi everyone,

I installed Gazebo8 via `brew`, and Bullet is installed along as a [dependency](https://github.com/osrf/homebrew-simulation/blob/master/gazebo8.rb#L41) with [`--with-shared`](https://github.com/Homebrew/homebrew-core/blob/master/Formula/bullet.rb#L36) and [`--with-double-precision`](https://github.com/Homebrew/homebrew-core/blob/master/Formula/bullet.rb#L45) parameters.

The whole installation process runs out smoothly without errors.
However, running `gazebo`, the following error is raised:
~~~bash
dyld: Library not loaded: @rpath/libBulletDynamics.2.85.dylib
  Referenced from: /usr/local/opt/bullet/lib/libBulletSoftBody.2.85.dylib
  Reason: image not found
Abort trap: 6
~~~

Running `otool -L /usr/local/opt/bullet/lib/libBulletSoftBody.2.85.dylib` reports
~~~bash
/usr/local/opt/bullet/lib/libBulletSoftBody.2.85.dylib:
	/usr/local/opt/bullet/lib/libBulletSoftBody.2.85.dylib (compatibility version 2.85.0, current version 2.85.0)
	@rpath/libBulletDynamics.2.85.dylib (compatibility version 2.85.0, current version 2.85.0)
	@rpath/libBulletCollision.2.85.dylib (compatibility version 2.85.0, current version 2.85.0)
	@rpath/libLinearMath.2.85.dylib (compatibility version 2.85.0, current version 2.85.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 307.4.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1238.0.0)
~~~
while running `otool -l /usr/local/opt/bullet/lib/libBulletSoftBody.2.85.dylib | grep -i rpath` reports
~~~bash
         name @rpath/libBulletDynamics.2.85.dylib (offset 24)
         name @rpath/libBulletCollision.2.85.dylib (offset 24)
         name @rpath/libLinearMath.2.85.dylib (offset 24)
~~~

With the latter command, it seems that `cmd LC_RPATH` load commands are missing.

I was able to workaround this problem by setting
~~~bash
export DYLD_FALLBACK_LIBRARY_PATH=$DYLD_FALLBACK_LIBRARY_PATH:/usr/local/opt/bullet/lib
~~~

erwincoumans, 2017-02-02:
do you have a suggested patch? Otherwise let's close the issue. Note that Gazebo doesn't use Bullet btMultiBody, so for robotics your are better of using pybullet.

claudiofantacci, 2017-02-02:
Hi @erwincoumans , thanks for reaching out.

I think that there is a problem with RPATH and bullet3.
It seems that bullet3 [enables RPATH](https://github.com/bulletphysics/bullet3/blob/master/CMakeLists.txt#L13-L14) but LC_RPATH are not actually set.
I know that in `brew` RPATH should be disabled, but there might be some problems because of a known issue/bug with CMAKE and macOS RPATH that some of my colleagues and me spotted recently: we opened an issue on KitWare GitLab [here](https://gitlab.kitware.com/cmake/cmake/issues/16589#note_221333).

I don't have a suggested patch/strategy for now. We first should wait for the CMake issue to be solved first, otherwise I might think of a possible workaround for the meantime.
To be honest, I should first download, compile and install bullet3 from source instead using the one provided by `brew` to see where the real problem could lie...

My setup comes from `brew`, and bullet is installed as a dependency by Gazebo. I don't think I can do much about it using `brew` setup.

claudiofantacci, 2017-06-21:
This post is an update about this issue.

It seems that this issue has been "**solved**" with PR #954.
The double quotes are because if I issue
```shell
~ $ otool -L /usr/local/opt/bullet/lib/libBulletSoftBody.2.86.dylib
/usr/local/opt/bullet/lib/libBulletSoftBody.2.86.dylib:
	/usr/local/opt/bullet/lib/libBulletSoftBody.2.86.dylib (compatibility version 2.86.0, current version 2.86.0)
	@rpath/libBulletDynamics.2.86.dylib (compatibility version 2.86.0, current version 2.86.0)
	@rpath/libBulletCollision.2.86.dylib (compatibility version 2.86.0, current version 2.86.0)
	@rpath/libLinearMath.2.86.dylib (compatibility version 2.86.0, current version 2.86.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 307.4.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1238.0.0)

~ $ otool -l /usr/local/opt/bullet/lib/libBulletSoftBody.2.86.dylib | grep -i rpath
         name @rpath/libBulletDynamics.2.86.dylib (offset 24)
         name @rpath/libBulletCollision.2.86.dylib (offset 24)
         name @rpath/libLinearMath.2.86.dylib (offset 24)
```
I can still find `@rpath` prefix which should not be there.

This problem may be due to the above mentioned problem on CMake which is fixed in the upcoming 3.9 version. Let's wait for the new release and see whether we can fix this definitely or not 😄 .

closed 2021-04-26 by erwincoumans