bulletphysics/bullet3 #924: ConvexPolyhedron assert issue.

ryanbartley, 2017-01-19:
Not sure if this is an issue but I can't seem to get passed it. I'm trying to compute a `btConvexHullShape` from a cloud of points and I keep running into an assert in btConvexPolyhedron.cpp, line 143, `btAssert(edptr->m_face1<0);`. This value is obviously not negative as it continues to assert there. I unfortunately don't totally understand what's going on here to fix it. Is there an initial step that i should take to clear out my point cloud? Any help would be greatly appreciated. For reference, I am using this code that I pulled from one of the examples. I can also make the point data available. Thanks again.

```
    auto shape = new btConvexHullShape;
    for( auto &vert : verts ) {
        shape->addPoint(vert, false);
    }
    shape->recalcLocalAabb();
    shape->optimizeConvexHull();
```

ryanbartley, 2017-01-19:
Moving this to the forum. Sorry about that.

closed 2017-01-19 by ryanbartley

erwincoumans, 2017-01-19:
If you can reproduce it in the example browser, please share a reproduction
case as issue (with points).

Thanks!
Erwin

ryanbartley, 2017-01-22:
In actuality, it was a submodule issue. The above code was written to correct a problem that I was getting with this code...
```
    auto shape = new btConvexHullShape( verts.data(), verts.size() );
    shape->initializePolyhedralFeatures()
```
...which was giving me the error mentioned above. I read on a forum post that the polyhedral features were still experimental. So I removed it in favor of the code in the original issue post above. I wrote the issue because it had crept back into my coworkers codebase. Once we found out that it was a stale submodule problem, it went away. Thanks so much for getting back to me so fast.