bulletphysics/bullet3 #935: Collision code for sphere vs triangles broken on November 22

olegklimov, 2017-01-26:
Commit 0516d2ecaaceb5ad43d8612413c676961e8e223d has broken my simulation.

My robot has sphere wheels. When moving over static triangle mesh, robot behaves erratically, effects range from bumping into something invisible to flyaways to infinity.

git checkout 24c9020aac15ba1278a33e20f8a753d519923a15   # no bug

git checkout 0516d2ecaaceb5ad43d8612413c676961e8e223d  # bug

Here is diff in question:

```diff
0516d2ecaaceb5ad43d8612413c676961e8e223d
 examples/pybullet/testrender.py                                        | 3 ++-
 src/BulletCollision/CollisionDispatch/SphereTriangleDetector.cpp       | 2 +-
 src/BulletCollision/CollisionDispatch/btCompoundCollisionAlgorithm.cpp | 3 +++
 .../CollisionDispatch/btSphereTriangleCollisionAlgorithm.cpp           | 2 +-
 4 files changed, 7 insertions(+), 3 deletions(-)

diff --git a/examples/pybullet/testrender.py b/examples/pybullet/testrender.py
index 5935c38..0d3475c 100644
--- a/examples/pybullet/testrender.py
+++ b/examples/pybullet/testrender.py
@@ -19,6 +19,7 @@ pixelHeight = 240
 nearPlane = 0.01
 farPlane = 1000
 lightDirection = [0,1,0]
+lightColor = [1,1,1]#optional argument
 fov = 60
 
 #img_arr = pybullet.renderImage(pixelWidth, pixelHeight)
@@ -28,7 +29,7 @@ for pitch in range (0,360,10) :
     viewMatrix = pybullet.computeViewMatrixFromYawPitchRoll(camTargetPos, camDistance, yaw, pitch, roll, upAxisIndex)
     aspect = pixelWidth / pixelHeight;
     projectionMatrix = pybullet.computeProjectionMatrixFOV(fov, aspect, nearPlane, farPlane);
-    img_arr = pybullet.getCameraImage(pixelWidth, pixelHeight, viewMatrix,projectionMatrix, lightDirection)
+    img_arr = pybullet.getCameraImage(pixelWidth, pixelHeight, viewMatrix,projectionMatrix, lightDirection,lightColor)
     w=img_arr[0]
     h=img_arr[1]
     rgb=img_arr[2]
diff --git a/src/BulletCollision/CollisionDispatch/SphereTriangleDetector.cpp b/src/BulletCollision/CollisionDispatch/SphereTriangleDetector.cpp
index 6340178..006cc65 100644
--- a/src/BulletCollision/CollisionDispatch/SphereTriangleDetector.cpp
+++ b/src/BulletCollision/CollisionDispatch/SphereTriangleDetector.cpp
@@ -100,7 +100,7 @@ bool SphereTriangleDetector::collide(const btVector3& sphereCenter,btVector3 &po
 	btScalar radiusWithThreshold = radius + contactBreakingThreshold;
 
 	btVector3 normal = (vertices[1]-vertices[0]).cross(vertices[2]-vertices[0]);
-	normal.normalize();
+	normal.safeNormalize();
 	btVector3 p1ToCentre = sphereCenter - vertices[0];
 	btScalar distanceFromPlane = p1ToCentre.dot(normal);
 
diff --git a/src/BulletCollision/CollisionDispatch/btCompoundCollisionAlgorithm.cpp b/src/BulletCollision/CollisionDispatch/btCompoundCollisionAlgorithm.cpp
index 1fc960d..7f4dea1 100644
--- a/src/BulletCollision/CollisionDispatch/btCompoundCollisionAlgorithm.cpp
+++ b/src/BulletCollision/CollisionDispatch/btCompoundCollisionAlgorithm.cpp
@@ -292,6 +292,9 @@ void btCompoundCollisionAlgorithm::processCollision (const btCollisionObjectWrap
 		btTransform otherInCompoundSpace;
 		otherInCompoundSpace = colObjWrap->getWorldTransform().inverse() * otherObjWrap->getWorldTransform();
 		otherObjWrap->getCollisionShape()->getAabb(otherInCompoundSpace,localAabbMin,localAabbMax);
+		btVector3 extraExtends(resultOut->m_closestPointDistanceThreshold, resultOut->m_closestPointDistanceThreshold, resultOut->m_closestPointDistanceThreshold);
+		localAabbMin -= extraExtends;
+		localAabbMax += extraExtends;
 
 		const ATTRIBUTE_ALIGNED16(btDbvtVolume)	bounds=btDbvtVolume::FromMM(localAabbMin,localAabbMax);
 		//process all children, that overlap with  the given AABB bounds
diff --git a/src/BulletCollision/CollisionDispatch/btSphereTriangleCollisionAlgorithm.cpp b/src/BulletCollision/CollisionDispatch/btSphereTriangleCollisionAlgorithm.cpp
index 280a4d3..86d4e74 100644
--- a/src/BulletCollision/CollisionDispatch/btSphereTriangleCollisionAlgorithm.cpp
+++ b/src/BulletCollision/CollisionDispatch/btSphereTriangleCollisionAlgorithm.cpp
@@ -56,7 +56,7 @@ void btSphereTriangleCollisionAlgorithm::processCollision (const btCollisionObje
 	
 	/// report a contact. internally this will be kept persistent, and contact reduction is done
 	resultOut->setPersistentManifold(m_manifoldPtr);
-	SphereTriangleDetector detector(sphere,triangle, m_manifoldPtr->getContactBreakingThreshold());
+	SphereTriangleDetector detector(sphere,triangle, m_manifoldPtr->getContactBreakingThreshold()+ resultOut->m_closestPointDistanceThreshold);
 	
 	btDiscreteCollisionDetectorInterface::ClosestPointInput input;
 	input.m_maximumDistanceSquared = btScalar(BT_LARGE_FLOAT);///@todo: tighter bounds
```

erwincoumans, 2017-01-27:
Thanks for the report, we should fix this. Do you have some small reproduction case (some dummy mesh and sphere etc)?

olegklimov, 2017-01-28:
Hello Erwin. Of course you're right that I should have put more effort on exploring the case myself. I have now, it turned out this changes have exposed my own bug -- I have empty triangles in my mesh.

Empty triangles was safely ignored *before*, but cause erratic behavior *after* this commit.

A kind of triangles that cause trouble: `p1 p2 p3` triangle with `p1==p2` and `p3` away, along the edge of a valid triangle.

I guess the most sensible thing to do is to print a warning (or assert, or ignore empty triangles again).

Here's a demo:

https://youtu.be/cf8c4aqnPdo

(small wheels at front and back are decorative, big wheels modeled as rotating spheres, base of the robot is restricted to be straight up)

olegklimov, 2017-01-28:
I mean, my problem is solved as I removed empty triangles from my simulation. Please feel free to close this. If you think it's worth doing changes, I'll be happy to test.

erwincoumans, 2017-01-28:
Thanks for getting back to this. I'm still interesting in a small repro case (some triangle mesh and a sphere rolling over it). Then we should discard degenerate triangles.
That looks great, is that the Fetch robot? If so, which URDF/SDF are you using exactly?

olegklimov, 2017-01-28:
https://github.com/olegklimov/bullet3/tree/issue935

Please note that all spheres affected, other primitives and convex bodies are not.

olegklimov, 2017-01-28:
> which URDF/SDF are you using exactly

It is modified version of original fetch_description. Modifications: 
* no bellows_link
* moments of inertia r_gripper_finger_link and l_gripper_finger_link set to non-zero
* disable rotation other than around vertical axis for base
* added camera
* spheres collision shape for wheels
* boxes for fingers collision shape

You can see more of Bullet problems in this video: https://youtu.be/SWJKh-DFzdE

erwincoumans, 2017-01-28:
Great, thanks a lot for the repro, I'll check it out. The contact/friction stability problems in the other video can be fixed by tuning various settings, we are actively working on exactly those scenarios. Is that modified URDF available somewhere, and the frying pan model?

erwincoumans, 2017-02-02:
Thanks for the report, please let us know if the fix didn't help.

closed 2017-02-02 by erwincoumans