bulletphysics/bullet3 #957: pybullet retrieving link velocities

benelot, 2017-02-11:
Can pybullet retrieve the speed of different multibody links? I could only find the getBaseVelocity method to retrtieve the base velocity. However, for many motion related RL applications, it is helpful to have all velocities. I am working with a VelocityHelper for the moment, which calculates the velocity by using the position difference, but I am not sure how well this will work. Are there better options?

erwincoumans, 2017-02-12:
Why did you close it? The joint velocity (scalar) is available, but not in Cartesian / World coordinates.

benelot, 2017-02-12:
I thought it is maybe a better question to ask on the forum. I may have missed something. And since this here is not a question/answer section of bullet, I thought I would come back and track it only after being sure that it is not available at all.

benelot, 2017-02-12:
I saw the joint velocity and this is at least something to start with. However, when thinking how I would have to calculate the Cartesian velocity, I thought it could be an unwanted obstacle for RL to learn this transformation. That is why I wanted to create this feature by myself. I would propose to create an interface for positions and velocities that is a bit more model agnostic (pretty sure you have that on your plan). I currently have methods that do exactly that:

closed 2017-05-28 by benelot