bulletphysics/bullet3 #959: pybullet getJointInfo missing joint limits

benelot, 2017-02-12:
I am currently working on a Motion Environment for OpenAI gym which supports reinforcement learning of different morphologies to perform motion tasks. The environment can work with any kind of robot by querying for joint number, type and limits to configure the action space for the agent. Joint number and type can easily be found, but the limits are not queryable. Or am I missing anything?

erwincoumans, 2017-02-13:
Not all data from URDF files is exposed at the moment, you can't query (nor change) most data...

benelot, 2017-02-13:
Maybe I can track it down and see how to add it. Would you be okay with adding it as another return value to getJointInfo?

benelot, 2017-02-13:
After some code diving, I realized that the limit get lost in ConvertURDF2Bullet as they are stored in MultiBodyCreationInterface and get instantiated as limits, but are then not stored anywhere...I thought maybe it is a quickwin, but it needs more thinking where to store the limits. If you want to give me a heads up, I will try. Otherwise yeah, I read it from URDF.

benelot, 2017-02-14:
I am trying to implement the return of joint limits with getJointInfo, unfortunately it segfaults for now. How do you debug pybullet usually? Currently I am stuck with printfs and such.

benelot, 2017-02-18:
How to run makesdna? Is there a script to do it? I need to regenerate the bullet.h and probably other files to make my getJointInfo(...) implementation with jointLowerLimit and jointUpperLimit work.

benelot, 2017-02-18:
Ok, I realized that it is harder to do than I thought. It seems that I need to regenerate bullet.h to add this functionality. Whenever you have time, maybe you can reuse my work.

erwincoumans, 2017-03-27:
I created a little document for the steps I take:
https://docs.google.com/document/d/1QiEXTx49Rk8DQKUijAmWEdI1wxV27ItNCCJoiSFw2ig/edit?usp=sharing
Unless you are really confident you are familiar with the serialization process, it is generally best not to touch the serialization structures and just report an issue in this tracker instead.

closed 2017-05-08 by benelot