bulletphysics/bullet3 #960: Building pybullet on macOS Sierra 10.12.3 w/ Numpy and Anaconda Python 3.5

thomasantony, 2017-02-14:
I used the following in `build_and_run_cmake_pybullet_double.sh`

```
cmake -DBUILD_PYBULLET=ON -DBUILD_PYBULLET_NUMPY=ON -DUSE_DOUBLE_PRECISION=ON - \
      -DCMAKE_BUILD_TYPE=Release \
      -DPYTHON_EXECUTABLE:FILEPATH=/Users/tantony/anaconda/bin/python3.5 \
      -DPYTHON_INCLUDE_DIR:PATH=/Users/tantony/anaconda/include/python3.5m \
      -DPYTHON_NUMPY_INCLUDE_DIR:PATH=/Users/tantony/anaconda/lib/python3.5/site-packages/numpy/core/include \
      -DPYTHON_LIBRARY:FILEPATH=/Users/tantony/anaconda/lib/libpython3.5m.dylib ..
```

to try and build pybullet with Numpy support. I have numpy installed using Anaconda. However, I kept getting errors in the linking phase such as:
```Undefined symbols for architecture x86_64:
  "_PyCObject_AsVoidPtr", referenced from:
      _initpybullet in pybullet.c.o
  "_PyCObject_Type", referenced from:
      _initpybullet in pybullet.c.o
  "_PyInt_FromLong", referenced from:
      _pybullet_connectPhysicsServer in pybullet.c.o
      _pybullet_loadSDF in pybullet.c.o
      _pybullet_loadBullet in pybullet.c.o
      _pybullet_loadMJCF in pybullet.c.o
      _pybullet_getNumBodies in pybullet.c.o
      _pybullet_getBodyUniqueId in pybullet.c.o
      _pybullet_getNumConstraints in pybullet.c.o
      ...
  "_PyString_FromString", referenced from:
      _pybullet_getBodyInfo in pybullet.c.o
      _pybullet_getJointInfo in pybullet.c.o
      _pybullet_getVisualShapeData in pybullet.c.o
  "_Py_InitModule4_64", referenced from:
      _initpybullet in pybullet.c.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)```

In order to fix this, I had to modify `examples/pybullet/CMakeLists.txt` and insert the following at line 88:
```
LINK_LIBRARIES(
        python
    )
```

Is this something specific to my system? If not, I can make a PR for this.

tjacobs, 2017-04-11:
Same thing with me, but it happens when I import the library. 

`
Thomass-Air:betabot thomasjacobs$ sudo python bot.py
Traceback (most recent call last):
  File "bot.py", line 20, in <module>
    from simulator import Simulator
  File "sim/simulator.py", line 1, in <module>
    import pybullet as p
ImportError: dynamic module does not define module export function (PyInit_pybullet)
Thomass-Air:betabot thomasjacobs$ python --version
Python 3.5.2 :: Continuum Analytics, Inc.
`

It happens when I run it with python 3.5.2, but it loads and runs fine with python 2.7. Any easy fix or workaround for 3.5?

erwincoumans, 2017-04-11:
Mac OSX Python header locations are messed up with framework, and Anaconda different locations (#include <Python/Python.h> versus #include <Python.h>)
This fixes it: https://github.com/bulletphysics/bullet3/pull/1068

Here are 3 easy options to install pybullet, with Anaconda:
In the root of the Bullet Physics SDK source tree, run
CFLAGS="-DB3_NO_PYTHON_FRAMEWORK" python setup.py install

cmake, add the flag -DBUILD_PYBULLET_MAC_USE_PYTHON_FRAMEWORK=OFF

Globally, just use pip or pip3:
CFLAGS="-DB3_NO_PYTHON_FRAMEWORK" pip install pybullet

closed 2017-04-11 by erwincoumans

erwincoumans, 2017-04-11:
Using latest Anaconda, this worked for me:

cmake -DBUILD_PYBULLET=ON -DBUILD_PYBULLET_NUMPY=OFF -DUSE_DOUBLE_PRECISION=ON -DCMAKE_BUILD_TYPE=Release -DBUILD_PYBULLET_MAC_USE_PYTHON_FRAMEWORK=OFF -DBUILD_PYBULLET_SHOW_PY_VERSION=ON -DPYTHON_INCLUDE_DIR=/Users/erwincoumans/anaconda3/include/python3.6m -DPYTHON_LIBRARY=/Users/erwincoumans/anaconda3/lib/libpython3.6m.dylib ..

make -j4
cd examples/pybullet
ln -s pybullet.dylib pybullet.so

Note that the optional cmake option  -DBUILD_PYBULLET_SHOW_PY_VERSION=ON  will show some warning with the actual PY_MAJOR_VERSION and PY_MINOR_VERSION during compilation:

Scanning dependencies of target pybullet
[ 80%] Building C object examples/pybullet/CMakeFiles/pybullet.dir/pybullet.o
/Users/erwincoumans/dev/bullet3/examples/pybullet/pybullet.c:23:9: warning: PY_MAJOR_VERSION=3 [-W#pragma-messages]
#pragma message(B3_VAR_NAME_VALUE(PY_MAJOR_VERSION))
        ^
/Users/erwincoumans/dev/bullet3/examples/pybullet/pybullet.c:24:9: warning: PY_MINOR_VERSION=6 [-W#pragma-messages]
#pragma message(B3_VAR_NAME_VALUE(PY_MINOR_VERSION))
        ^