bulletphysics/bullet3 #976: Concerning NaNs in pybullet

benelot, 2017-02-26:
I just had a RL agent trick me several times by sending NaN into my pybullet controlled simulation. I had this problem before with bullet physics that the rest of my simulator somehow produced NaNs and set them for positions or similar, which messed up the physics simulation. Given that people using pybullet may not be aware of such an issue, maybe it could help to check if values are numerical values (e.g. not NaN).

erwincoumans, 2017-02-26:
Well, I would expect that if you put NaN into any system, you won't get a results better than NaN :)
Perhaps you add some check in python? (math.isnan())?

benelot, 2017-02-26:
I fixed it with exactly that, I just thought maybe somebody might miss to
catch that and so pybullet could notify the user of this. I just wanted to
point this out that this is a hidden source of error.

ktfh, 2017-02-27:
A single function to reset a rigid body's states would be nice, I think if a NaN or Inf creeps in by accident there is a lot of state to fix. I wrote some bad code that applied Inf or NaN forces and the procedure to fix a rigid body was like; setWorldTransform, forceActivationState, clearForces, setLinearVelocity, setAngularVelocity, updateInertiaTensor.

erwincoumans, 2017-02-27:
You really should never pass NaN as arguments, it will potentially corrupt more than just a single rigid body.

benelot, 2017-02-28:
It is probably better to catch them and pull them to zero or whatever works best in your case. In RL, I punish the unwanted by setting the reward to 0 and by terminating the episode. That will teach the agent not to send this as it seems to be a terminate/fail signal to send.

closed 2017-03-04 by benelot

kartikpaigwar, 2021-02-15:
> I just had a RL agent trick me several times by sending NaN into my pybullet controlled simulation. I had this problem before with bullet physics that the rest of my simulator somehow produced NaNs [...]

I am also finding this issue, GetBasePositionOrientation( ) is returning NaN during the training. I am training robot on custom heighfield map. What might be the reason for getting NaNs, my RL agent is not changing the simulation parameters to NaN.

erwincoumans, 2021-02-15:
Are you clamping the data you send to PyBullet in the range [-1,1] ? If you can reproduce it in a small case, please share the code.