bulletphysics/bullet3 #984: Loading/connecting to VR demo map

nimaasg, 2017-03-03:
@benelot @erwincoumans
I'm trying to work with the VR demo map, but the following two ways didn't work for me:

* Trying to load the map via pybullet when connected to Physics Server (running in ExampleBrowser). According to [this](http://bulletphysics.org/mediawiki-1.5.8/index.php/Using_pybullet) issuing p.setRealTimeSimulation(1) should load the VR map, but it doesn't do anything. Is there another way of explicitly loading the entire map?

* Starting the VR demo executable (App_SharedMemoryPhysics_VR_vs2010) and trying to connect to it via pybullet. I get this error:
>>> p.connect(p.SHARED_MEMORY)
b3Warning[C:\Users\Nima\Documents\Robotics\Bullet\bullet3-2.86\examples\SharedMemory\Win32SharedMemory.cpp,99]:
Could not create file mapping object (2).
b3Warning[C:\Users\Nima\Documents\Robotics\Bullet\bullet3-2.86\examples\SharedMemory\PhysicsClientSharedMemory.cpp,258]:
Cannot connect to shared memory-1

Any thoughts about these issues?

Thanks!

erwincoumans, 2017-03-03:
You need to start the standalone physics server and use the --robotassets flag.
Alternatively, you can simply run the bullet3\examples\pybullet\vr_kuka_setup.py script
>>to connect to it via pybullet. I get this error:

You need to use a matching version of physics server (VR, GUI, example browser etc) and pybullet build. If that still fails, reboot your machine and make sure to start the physics server before any client (pybullet etc)

nimaasg, 2017-03-04:
Thanks @erwincoumans.

> You need to start the standalone physics server and use the --robotassets flag.

By standalone do you mean NOT the physics server from within example browser? How do I invoke the standalone physics server?

> You need to use a matching version of physics server 

Makes sense. I was using the executable downloaded from bullet repo. How can I build the VR executable from my installation of bullet?

erwincoumans, 2017-03-04:
On Windows, make sure Python is installed in C:\python* or modify
build_visual_studio_vr_pybullet_double.bat
and click build_visual_studio_vr_pybullet_double.bat
You could also click build_visual_studio_vr_pybullet_double_cmake.bat, but I don't recommend using cmake, the other link is better (build_visual_studio_vr_pybullet_double.bat).

Premake creates those 2 binaries for standalone physics servers:
App_PhysicsServer_SharedMemory_GUI and
App_PhysicsServer_SharedMemory_VR

Please let me know if that helps.

nimaasg, 2017-03-10:
Thanks @erwincoumans, this actually worked and I'm able to use App_PhysicsServer_SharedMemory_VR with pybullet smoothly. The key was to use the master source version.

If I want to get the standalone physics server built on Mac OS to work with pybullet, what's the best way to do that?

Utsavz, 2018-09-26:
Hi, @nimaasg could you please list all the steps that made it work for you.?
How do I start a physics server for vr using bullet3 and how do I connect to it using pybullet

erwincoumans, 2019-12-20:
Closing issue. Just compile and build App_PhysicsServer_SharedMemory_VR  and run it. Then connect with pybullet using SHARED_MEMORY connection mode.
Mac OS doesn't support VR properly.

closed 2019-12-20 by erwincoumans