bulletphysics/bullet3 #989: btDefaultSerializer: 'btHashPtr': no appropriate default constructor available

nicolaichuk, 2017-03-05:
In btDefaultSerializer used btHashPtr for call function:

```
void	btAlignedObjectArray::resize(int newsize, const T& fillData=T())
```
but btHashPtr do not have default constructor.

```
linearmath\btalignedobjectarray.h(218): error 'btHashPtr': no appropriate default constructor available

linearmath\btalignedobjectarray.h(218): note: No constructor could take the source type, or constructor overload resolution was ambiguous

linearmath\btalignedobjectarray.h(504): note: while compiling class template member function 'void btAlignedObjectArray<Key>::copyFromArray(const btAlignedObjectArray<Key> &)'
1>          with
1>          [
1>              Key=btHashPtr
1>          ]

linearmath\btalignedobjectarray.h(67): note: see reference to function template instantiation 'void btAlignedObjectArray<Key>::copyFromArray(const btAlignedObjectArray<Key> &)' being compiled
1>          with
1>          [
1>              Key=btHashPtr
1>          ]

linearmath\bthashmap.h(228): note: see reference to class template instantiation 'btAlignedObjectArray<Key>' being compiled
1>          with
1>          [
1>              Key=btHashPtr
1>          ]

linearmath\btserializer.h(180): note: see reference to class template instantiation 'btHashMap<btHashPtr,void *>' being compiled
```

erwincoumans, 2017-03-05:
Thanks for the report. What compiler + version + platform reports this error?

nicolaichuk, 2017-03-05:
https://github.com/bulletphysics/bullet3/tree/2.86.1

windows 7, msvc 2015 x64

erwincoumans, 2017-03-05:
That's odd, I'm using MSVC 2015, 64bit all the time, without errors. Are you compiling the Bullet sdk/examples using premake4-generated project files or cmake-generation project files, or your own files/build system? Could it be the some compiler flag?

nicolaichuk, 2017-03-05:
I build swig generate code for java jni.
It generate operator= for this public field

```
btHashMap<btHashPtr,void*> m_skipPointers;
```
https://github.com/bulletphysics/bullet3/blob/master/src/LinearMath/btSerializer.h#L390

Now i did this field immutable in swig config and all build ok.

P.S.: Public access field in class is a bad practice, imho.

closed 2017-03-05 by nicolaichuk