bulletphysics/bullet3 #1001: indefine reference when compiling for Android

kiuKisas, 2017-03-08:
Hi.
I try to compile the lib for an android/ios game, so this is what I done:

- clone the depot
- `cmake build3/Android/jni/Android.mk .` to generate a proper Makefile, I got this message btw:

```
OPENGL FOUND
/usr/lib64/libGLU.so/usr/lib64/libGL.so
CMake Warning (dev) in test/CMakeLists.txt:
  The source directory

    /home/equinox/code/bullet3/test/RobotLogging

  does not contain a CMakeLists.txt file.

  CMake does not support this case but it used to work accidentally and is
  being allowed for compatibility.

  Policy CMP0014 is not set: Input directories must have CMakeLists.txt.  Run
  "cmake --help-policy CMP0014" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/equinox/code/bullet3

```
So, I performed a simple `make` and I have all this dependencies issues:

```
[  1%] Built target LinearMath
[ 14%] Built target BulletCollision
[ 19%] Built target BulletDynamics
[ 19%] Built target App_HelloWorld
[ 20%] Built target App_BasicExample
[ 22%] Built target OpenGLWindow
[ 23%] Built target Bullet3Common
[ 23%] Built target AppBasicExampleGui
[ 24%] Built target BulletExampleBrowserLib
[ 33%] Built target gwen
[ 34%] Built target BussIK
[ 35%] Built target BulletInverseDynamicsUtils
[ 36%] Built target BulletInverseDynamics
[ 38%] Built target BulletSoftBody
[ 54%] Built target App_ExampleBrowser
[ 55%] Built target BulletFileLoader
[ 55%] Built target BulletWorldImporter
[ 55%] Linking CXX executable App_RobotSimulator
CMakeFiles/App_RobotSimulator.dir/__/MultiThreading/b3PosixThreadSupport.o : Dans la fonction « b3PosixThreadSupport::waitForResponse(int*, int*) » :
b3PosixThreadSupport.cpp:(.text+0x18) : référence indéfinie vers « sem_wait »
b3PosixThreadSupport.cpp:(.text+0x96) : référence indéfinie vers « sem_wait »
CMakeFiles/App_RobotSimulator.dir/__/MultiThreading/b3PosixThreadSupport.o : Dans la fonction « threadFunction(void*) » :
b3PosixThreadSupport.cpp:(.text+0xec) : référence indéfinie vers « sem_post »
b3PosixThreadSupport.cpp:(.text+0x102) : référence indéfinie vers « sem_wait »
b3PosixThreadSupport.cpp:(.text+0x116) : référence indéfinie vers « sem_wait »
b3PosixThreadSupport.cpp:(.text+0x14a) : référence indéfinie vers « sem_post »
b3PosixThreadSupport.cpp:(.text+0x15e) : référence indéfinie vers « sem_post »
b3PosixThreadSupport.cpp:(.text+0x1ac) : référence indéfinie vers « sem_post »
CMakeFiles/App_RobotSimulator.dir/__/MultiThreading/b3PosixThreadSupport.o : Dans la fonction « b3PosixThreadSupport::runTask(int, void*, int) » :
b3PosixThreadSupport.cpp:(.text+0x23c) : référence indéfinie vers « sem_post »
b3PosixThreadSupport.cpp:(.text+0x25c) : référence indéfinie vers « sem_post »
CMakeFiles/App_RobotSimulator.dir/__/MultiThreading/b3PosixThreadSupport.o : Dans la fonction « b3PosixThreadSupport::isTaskCompleted(int*, int*, int) » :
b3PosixThreadSupport.cpp:(.text+0x2a2) : référence indéfinie vers « sem_trywait »
CMakeFiles/App_RobotSimulator.dir/__/MultiThreading/b3PosixThreadSupport.o : Dans la fonction « b3PosixThreadSupport::createBarrier() » :
b3PosixThreadSupport.cpp:(.text+0x360) : référence indéfinie vers « pthread_barrier_init »
CMakeFiles/App_RobotSimulator.dir/__/MultiThreading/b3PosixThreadSupport.o : Dans la fonction « b3PosixThreadSupport::deleteBarrier(b3Barrier*) » :
b3PosixThreadSupport.cpp:(.text+0x3b4) : référence indéfinie vers « pthread_barrier_destroy »
CMakeFiles/App_RobotSimulator.dir/__/MultiThreading/b3PosixThreadSupport.o : Dans la fonction « b3PosixThreadSupport::stopThreads() » :
[... 621 lines cut ...]
glew.c:(.text+0x16bbb) : référence indéfinie vers « dlsym »
glew.c:(.text+0x16bdb) : référence indéfinie vers « dlopen »
glew.c:(.text+0x16bfb) : référence indéfinie vers « dlsym »
../OpenGLWindow/libOpenGLWindow.a(glew.o) : Dans la fonction « dlglXGetProcAddressARB » :
glew.c:(.text+0x97bd) : référence indéfinie vers « dlsym »
../OpenGLWindow/libOpenGLWindow.a(X11OpenGLWindow.o) : Dans la fonction « X11OpenGLWindow::X11OpenGLWindow() » :
X11OpenGLWindow.cpp:(.text+0x464) : référence indéfinie vers « dlopen »
X11OpenGLWindow.cpp:(.text+0x481) : référence indéfinie vers « dlsym »
X11OpenGLWindow.cpp:(.text+0x4a2) : référence indéfinie vers « dlsym »
X11OpenGLWindow.cpp:(.text+0x4c3) : référence indéfinie vers « dlsym »
X11OpenGLWindow.cpp:(.text+0x4e4) : référence indéfinie vers « dlsym »
X11OpenGLWindow.cpp:(.text+0x505) : référence indéfinie vers « dlsym »
../OpenGLWindow/libOpenGLWindow.a(X11OpenGLWindow.o):X11OpenGLWindow.cpp:(.text+0x526) : encore plus de références indéfinies suivent vers « dlsym »
../OpenGLWindow/libOpenGLWindow.a(X11OpenGLWindow.o) : Dans la fonction « X11OpenGLWindow::enableOpenGL() » :
X11OpenGLWindow.cpp:(.text+0xadb) : référence indéfinie vers « pthread_getconcurrency »
collect2: erreur : ld a retourné 1 code d'état d'exécution
make[2]: *** [examples/RobotSimulator/CMakeFiles/App_RobotSimulator.dir/build.make:1720: examples/RobotSimulator/App_RobotSimulator-2.86] Error 1
make[1]: *** [CMakeFiles/Makefile2:515: examples/RobotSimulator/CMakeFiles/App_RobotSimulator.dir/all] Error 2
make: *** [Makefile:139: all] Error 2

```

I try to add `-lpthread -ltr` options to `install/strip` and `install/local` but nothing change.

Thank's for your time !

erwincoumans, 2017-03-09:
Can you also try to add -ldl (to get rid of dlsym, dlopen undefined references)?

kiuKisas, 2017-03-09:
Nothing change.. I attached the generate Makefile with my modification if that can help you

erwincoumans, 2017-03-09:
You may have more luck in Android forums. Good luck!

closed 2017-03-09 by erwincoumans