bulletphysics/bullet3 #997: Lost implementation "int btCollisionWorldImporter::getNumConstraints() const"

nicolaichuk, 2017-03-08:
Function [int btCollisionWorldImporter::getNumConstraints() const](https://github.com/bulletphysics/bullet3/blob/master/src/BulletCollision/CollisionDispatch/btCollisionWorldImporter.h#L127)
do not have implementation in [cpp-file](https://github.com/bulletphysics/bullet3/blob/master/src/BulletCollision/CollisionDispatch/btCollisionWorldImporter.cpp).

nicolaichuk, 2017-03-08:
I think it is copy-past from https://github.com/bulletphysics/bullet3/blob/master/Extras/Serialize/BulletWorldImporter/btWorldImporter.h#L139

and this function may remove because collision word do not have constraints.

erwincoumans, 2017-03-08:
Yes, we can remove the method from the header file. Thanks for the report!

erwincoumans, 2017-03-08:
Fixed here:https://github.com/bulletphysics/bullet3/commit/644e510c83f9108f79d9ae98411e0a0f97be208b

closed 2017-03-08 by erwincoumans

nicolaichuk, 2017-03-08:
@erwincoumans, thanks.