bulletphysics/bullet3 #1007: Apply consistent formatting / code style

erwincoumans, 2017-03-10:
Contributors often use inconsistent code style (they use spaces instead of tabs,
open brackets at the end of line, instead of at the beginning etc).

Bullet code style guide is here:
https://docs.google.com/document/d/1u9vyzPtrVoVhYqQOGNWUgjRbfwfCdIts_NzmvgiJ144/edit

One option would be to create a style file for clang format.
https://clang.llvm.org/docs/ClangFormatStyleOptions.html

erwincoumans, 2017-03-13:
See https://github.com/bulletphysics/bullet3/pull/1011
Created a _clang-format file, starting with Google style, then modified:
ColumnLimit:     0
UseTab:          Always
ReflowComments:  false
SortIncludes:    false
May still need some tweaks, tried it on harmless examples/HelloWorld/HelloWorld.cpp

erwincoumans, 2017-03-13:
I'm not going to apply the clang file to the entire codebase at once, to avoid conflicts with pull requests etc. Thinking how to proceed, applying in most urgent files first.

AndresTraks, 2017-03-13:
I'd recommend:
UseTab: ForContinuationAndIndentation
PointerAlignment: Left

Tabs are useful for indentation (at the start of a line), because they look good at whatever width the tab character is configured to. But with tabs anywhere else on the line, there is a danger of messing up formatting if you want to align certain things across lines. For example, the variable names here are properly aligned in VS if tab width = 4, but not in GitHub, where tab width = 8:
https://github.com/bulletphysics/bullet3/blob/master/src/BulletDynamics/Featherstone/btMultiBody.h#L712

As per Google's style, I'd also make it a recommendation to avoid multiple declarations of pointers on one line. https://google.github.io/styleguide/cppguide.html#Pointer_and_Reference_Expressions

I've been hoping for formatting changes in Bullet for a long time. Thanks! :)

erwincoumans, 2017-03-13:
OK, I consider UseTab: ForContinuationAndIndentation.
We already have PointerAlignment: Left.

>>  I'd also make it a recommendation to avoid multiple declarations of pointers on one line.
I agree. Does it often happen in the current Bullet codebase?

Thanks for the feedback!

AndresTraks, 2017-03-29:
Ah, looks like DerivePointerAlignment: true overrides PointerAlignment. It analyzes the current style of the file and so btVector3.h still got & and * on the right. So I'd also recommend DerivePointerAlignment = false.

> Does it often happen in the current Bullet codebase?

Not really, a few times only. The main reason for this recommendation is that declarations like these are really confusing:
https://github.com/bulletphysics/bullet3/blob/79d24db60f2c916ffcdaf8b074be622e2e5bc609/src/BulletDynamics/MLCPSolvers/btDantzigLCP.cpp#L1358

But also, this would be consistent with PointerAlignment: Left, because the only reason you'd want to have & or * on the right is when defining multiple pointers on one line. To me, it makes sense that & and * are part of the type and therefore should be together with the type name.

erwincoumans, 2018-12-23:
We applied clang-format-all.sh a while ago. Closing issue.

closed 2018-12-23 by erwincoumans