bulletphysics/bullet3 #1014: Example Browser missing "Physics Server"

shazoom, 2017-03-14:
I was trying to follow the pybullet introduction [article](http://bulletphysics.org/mediawiki-1.5.8/index.php/Using_pybullet)  but when I run App_ExampleBrowser I don't have the option in the left pane for Physics Server under Experiments.

It's built on Mint 18.

erwincoumans, 2017-03-14:
It is moved out of Experiments into its own section, 3rd category from the top: Physics Server-Client. See screenshot.
[image]

closed 2017-03-14 by erwincoumans

benelot, 2017-03-14:
I will update the article soon, mistake of mine! In the article it refers
to the most recent version of it being in the google docs.

shazoom, 2017-03-15:
Thanks. I should have spotted that...but I didn't 😁

benelot, 2017-03-15:
I will add a last updated tag, so that people see how old it is. Also I
think the article becomes more stable now as the feature is no longer very
new and experimental.