bulletphysics/bullet3 #1022: btHeightfieldTerrainShape letting thin boxes through

erikogenvik, 2017-03-14:
I'm experiencing a strange behavior with btHeightfieldTerrainShape. It appears that boxes are slipping through the terrain, but only if they are elongated enough, and only in places (or with specific velocities).
I've made a little video showing the effect here: https://youtu.be/8DUJ1aoMd50
Square boxes aren't affected, but the more elongated the box becomes the more it seems to be.
CCD makes no difference (except for not allowing small square boxes falling completely through). As can be seen in the movie, the boxes seems to actually first collide with the terrain, only to then "slip" through.
Adding a margin to the height field has some effect, but will not be a guarantee.
I've also added a PR with a demo (the one shown in the movie) of the effect at https://github.com/bulletphysics/bullet3/pull/1021
I've yet to look closer at the actual collision detection code; is this a known issue perhaps?

gregeryb, 2017-04-27:
My guess is a ratio problem. Is that terrain one big quad? Or is it subdivided into smaller quads? Smaller quads give less of size ratio between the colliding quad and the box. Terrain height fields are suspect because they aren't sold, i.e. if in the real world the ground had width and breadth but was infinitely thin then things would fall through and computer simulations would be correct.

erwincoumans, 2017-04-27:
It is due to local penetration depth recovery, heightfield elements push the boxes sideways instead of upward. We should fix it at some stage (I have a lot of other things to do).

erikogenvik, 2017-04-27:
Thanks for the description. I'll look into replacing our terrain with triangle meshes instead. Unfortunately I'm not accustomed to the Bullet code enough to look into this myself.

BluePrintRandom, 2017-04-28:
What you really need is a dissolve quad faces to box, and triangles to
triangle box, and a method to cull the generated shapes by a tree or
something.

(dissolve triangle mesh and height field into their tree of shapes to test)

closed 2021-04-26 by erwincoumans