bulletphysics/bullet3 #1026: C API: torque joint control doesn't work for MJCF, SDF

olegklimov, 2017-03-18:
`loadMjcf()` and `loadSdf()` both call `processImportedObjects()`, which **creates motor for every joint**.

This is wrong, because if API user wants to uses `CONTROL_MODE_TORQUE`, he calls  `addJointTorqueMultiDof()`  that passes torque directly to Featherstone solver. Motor counteracts this torque in order to satisfy its uninitialized (zero) target and body **cannot move**.

TODO: enable motor only if `CONTROL_MODE_VELOCITY` or `CONTROL_MODE_POSITION_VELOCITY_PD` ever called for a joint.

olegklimov, 2017-03-18:
(not sure if it is possible to add motor later, after body has been created)

erwincoumans, 2017-03-18:
It is by design, see docs. You can disable the motor by setting velocity control with max force to 0.

olegklimov, 2017-03-18:
Right, it worked. I've edited the doc. But maybe set force to zero right in [PhysicsServerCommandProcessor.cpp:2700](https://github.com/bulletphysics/bullet3/blob/master/examples/SharedMemory/PhysicsServerCommandProcessor.cpp#L2770) ?

erwincoumans, 2018-12-23:
Things should work, let's close the old issue.

closed 2018-12-23 by erwincoumans