bulletphysics/bullet3 #1040: C API: b3LinkState lacks link velocity, rotational velocity

olegklimov, 2017-03-24:
```cpp
struct b3LinkState
{
	//m_worldPosition and m_worldOrientation of the Center Of Mass (COM)
    double m_worldPosition[3];
    double m_worldOrientation[4];

    double m_localInertialPosition[3];
    double m_localInertialOrientation[4];

	///world position and orientation of the (URDF) link frame
	double m_worldLinkFramePosition[3];
    double m_worldLinkFrameOrientation[4];
};
```
No way to know what world-relative speed vector of link is, also world-relative rotational velocity.

```cpp
struct SendActualStateArgs
{
	...
	double m_linkState[7*MAX_NUM_LINKS];
```

It is also not transmitted to client.

(top snippet also shows indentation problems still unfixed, impeding any progress)

erwincoumans, 2017-03-24:
That's right, no-one has asked for the world-frame velocity of links yet. It would be a good idea to add. Do you feel adventurous to add it? I did all the work to prepare for easy clang-format style re-formatting, but don't want to disturb development right now, let's wait a bit for that. Please don't reformat right now :)

olegklimov, 2017-03-24:
At least I have your blessing...

And I also probably need link mass (to be able to calculate com point, body impulse).

olegklimov, 2017-03-24:
Dup #957

erwincoumans, 2017-03-24:
I can add it in the next half hour :)

olegklimov, 2017-03-24:
That would be great. Also: joint limits, joint max effort, joint max speed.

benelot, 2017-03-24:
I think the static parts should go into getJointInfo, the others into the
JointState, am I wrong for some reason? Joint limits will not change
afterwards or am I wrong?

olegklimov, 2017-03-24:
> think the static parts should go into getJointInfo

You're not wrong here, and I guess your PR is done right.

erwincoumans, 2017-03-24:
> struct SendActualStateArgs
> {
> 	...
> 	double m_linkState[7*MAX_NUM_LINKS];
> It is also not transmitted to client.

This is the b3LinkState m_worldPosition/m_worldOrientation. Is that not transmitted?

olegklimov, 2017-03-24:
No no no speed is not transmitted, of course coordinates are there.

benelot, 2017-03-25:
I would love to have this information too. My MotionEnv is meant to learn to move with any kind of morphology but it requires to know a lot about the morphology to configure the agent correctly.

erwincoumans, 2017-03-26:
This pull request is going to take care of it:
https://github.com/bulletphysics/bullet3/pull/1043

Note that jointMaxForce/jointMaxVelocity is not processed in Bullet at the moment. The max force can be set as part of the setJointMotorControl2 command. For jointMaxVelocity we need to implement something in the constraints.

olegklimov, 2017-05-04:
IT WORKS

It really does.

(I wish we had an option to average speed over substeps -- it's a bit noisy now)

closed 2017-05-04 by olegklimov

erwincoumans, 2017-05-05:
The 'noise' may be due to too rigid contact. You can try making the floor a bit compliant:
When using URDF, you can add a contact section in the <link> add some
<contact>
      <lateral_friction value=".1"/>
      <stiffness value="300000"/>
      <damping value="1000"/>
    </contact>
MJCF xml likely has a similar stiffness/damping factor for contact that we can expose. That would remove a lot of 'jitter'/noise.