bulletphysics/bullet3 #1044: Some examples are broken on bulletphysics.org

cprogrammer1994, 2017-03-27:
On the [Collision_Callbacks_and_Triggers](http://www.bulletphysics.org/mediawiki-1.5.8/index.php?title=Collision_Callbacks_and_Triggers) page the following error occurres:

`Process 'highlight --fragment --syntax='cpp' --style-outfile='/tmp/highlight.css'' failed to start?`

Example code is not readable.

benelot, 2017-03-27:
True, something broke the wiki.

cprogrammer1994, 2017-03-29:
seems to work fine, thank you!

erwincoumans, 2017-03-31:
At some stage I'd like to retire the bulletphysics.org/wiki and move to use github.com/bulletphysics/bullet wiki. That way, I (or someone else) can review all content, make sure it doesn't conflict with existing documentation etc.

benelot, 2017-04-01:
Sounds like a good idea! I will start to port my articles soon and create
new ones there.

a-day-old-bagel, 2017-04-07:
So, was this going to get fixed? It's still broken on every page on the wiki for me, in both chrome and firefox. It makes it very hard to learn anything about bullet, since some of the topics don't appear to be in the manual (or am I wrong?)

benelot, 2017-04-07:
No, these things are not in the manual. I have no idea what might have
caused this. Did you do any changes on the server recently? Because it
seems that cpp as a language is gone on one hand and that changing the
language to something like c does not work either. The highlight command
seems to be broken.

Banderi, 2017-04-15:
This really is a bummer. Is there no really nothing that could fix the command, or could it be temporarily disabled to see the raw text?

benelot, 2017-04-15:
I can understand that it is very annoying, as it totally inhibits you from reading the code sections...

@Banderi: I can give you a list of those that make sense to be ported.

benelot, 2017-05-29:
It seems to work again.

closed 2021-04-26 by erwincoumans