bulletphysics/bullet3 #1056: setJointMotorControl2: Setting positions outside of the limits causes simulation issues

benelot, 2017-04-04:
I am using setJointMotorControl2 using a simple script I adapted from one of yours.

```
#!/usr/bin/env python

import pybullet as p
import time
import math
import numpy as np

useRealTime = 0
fixedTimeStep = 0.001

physId = p.connect(p.SHARED_MEMORY)
if (physId<0):
	p.connect(p.GUI)

p.loadURDF("ground.urdf")
p.setGravity(0,0,-9.81)
p.setTimeStep(fixedTimeStep)

p.setRealTimeSimulation(0)
# bodies = p.loadMJCF("mjcf/ant.xml")
# bodies = p.loadMJCF("mjcf/point.xml")
# bodies = p.loadMJCF("mjcf/half_cheetah.xml")
# bodies = p.loadMJCF("mjcf/humanoid.xml")
# bodies = p.loadMJCF("mjcf/humanoidstandup.xml")
# bodies = p.loadMJCF("mjcf/inverted_double_pendulum.xml")
# bodies = p.loadMJCF("mjcf/inverted_pendulum.xml")
# bodies = p.loadMJCF("mjcf/humanoid_symmetric.xml")
# bodies = p.loadMJCF("mjcf/hopper.xml")
# bodies = p.loadMJCF("mjcf/reacher.xml")
# bodies = p.loadMJCF("mjcf/swimmer.xml")
bodies = p.loadMJCF("mjcf/walker2d.xml")

minLimit = -math.pi/2
maxLimit = math.pi/2

p_gain = 0.8
speed = 1

# exclude = np.ones([len(bodies),50])
exclude = np.zeros([len(bodies),50])

exclude[1,4] = 0
exclude[1,5] = 0
exclude[1,8] = 1

for j in range(len(bodies)):
	for i in range(p.getNumJoints(bodies[j])):
		print "Setting [", j , "," , i , "]"
		p.setJointMotorControl2(bodies[j],i,p.TORQUE_CONTROL, 0)

t = 0
while True:
	print math.sin(t*speed)/2*(maxLimit-minLimit)
	for j in range(len(bodies)):
		for i in range(p.getNumJoints(bodies[j])):
			if exclude[j,i] == 1:
				print "Setting [", j , "," , i , "]"
				p.setJointMotorControl2(bodies[j],i,p.POSITION_CONTROL,math.sin(t*speed)/2*(maxLimit-minLimit),p_gain)
	p.stepSimulation()
	t = t + fixedTimeStep
```

Since not all joints are limited to -math.pi/2 math.pi/2, this will apply out of limits positions. Applying positions outside of the limits causes the whole multibody to fly around. Not sure if you want to do something about this. If I read the joint limits from file and apply only up to those, it worked very well (I experienced only some minor spasm as it overshoots the limit).

benelot, 2017-04-04:
Erwin Coumans wrote: 
> Yes, setJointMotorControl2 in conflict with a joint limit will cause issues. We could protect against that, by merging motor and limit in one constraint (like done in 6dof constraint).

benelot, 2018-01-06:
Fixed by simple clamping to limits in: commit c59a3763e5

erwincoumans, 2018-01-06:
Yes, I added clamping for now. Eventually I'll create a combined joint limit/motor for btMultiBody. The btGeneric6DofSpring2Constraint provides similar functionality for btRigidBody.

erwincoumans, 2018-12-23:
Clamping should deal with it, closing.

closed 2018-12-23 by erwincoumans