bulletphysics/bullet3 #1058: Proper API Documentation

ManuelTS, 2017-04-05:
Full API documentation for each single class, method, and field explaining its purpose, usage, parameters, requirements, and dependencies.

Rasons why can be found here:
 1) [Documenting your Code – Is it Important?](http://www.rapidvaluesolutions.com/documenting-your-code-is-it-important/)
 2) [What is the importance of technical documentation in software development?](https://www.quora.com/What-is-the-importance-of-technical-documentation-in-software-development)
 3) [On Importance of Documentation](https://www.dennisdel.com/on-importance-of-documentation/)
 4) [TLDR; or, The Importance of Documentation](http://www.wiliam.com.au/wiliam-blog/tldr-or-the-importance-of-documentation)

erwincoumans, 2017-04-05:
If you want to use Bullet with a documented API, go for pybullet: pybullet.org has a link to full documented API.

closed 2017-04-05 by erwincoumans

ManuelTS, 2017-04-05:
The [pybullet documentation](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.d6ihmmtes1id) is a lot more detailed than the [bullet API documentation](http://bulletphysics.org/Bullet/BulletFull/index.html), yes. But not an API documentation in the first place where all classes, methods, and fields and their purpose, usage, parameters, requirements, and dependencies are explained. Nevertheless, integrating the first doc into the second would be fruitful, since looking into the pybullet doc if I'm using c++ has a high strangeness.

erwincoumans, 2017-04-05:
pybullet doesn't have classes, only methods. All methods and their input arguments and output are explained. Since I am planning towards a future Bullet API, there are no plans to further document Bullet 2.x. You can use doxygen or look at the examples in the example browser.

Good luck!

benelot, 2017-04-05:
Or just ask on the bulletphysics.org forum, there are many active people
there!

benelot, 2017-04-05:
Just in case you are interested in making some documentation by providing more examples to the example browser, I can show you the way how you can do this easily.

ManuelTS, 2017-04-06:
Thanks for you replies and suggestions. I created this issue not to get workarounds to another pybullet doc or the forum but rather simply state the here in this very github repo the API doc is missing.

benelot, 2017-04-06:
I agree with you. When I started using Bullet, I was relying on all the
examples I found on the net and the examples in the repository. I feel too
that once there is a new Bullet API (3.x.x ++), there should be a better
API documentation, maybe some starter tutorials to get people interested
and some sample projects which show you how to build a small game, a small
physics simulator and a small whatsoever. I think in the longterm-view,
this also helps the refactoring of the code because more people start
contributing back to it. I do not know how Erwin thinks about this. Just my
2 c.

alanjfs, 2019-02-21:
This issue came up when searching for "doxygen". Didn't see it linked, and am not sure when it is from, but there is this:

- https://pybullet.org/Bullet/BulletFull/index.html

Which is better than none, but I'm missing (1) search functionality and (2) tree-view, similar to what PhysX has got.

- http://gameworksdocs.nvidia.com/PhysX/4.0/documentation/PhysXAPI/files/index.html

I'm not well versed with Doxygen, but was wondering whether it's perhaps a matter of simply rebuilding with a newer version of Doxygen?

erwincoumans, 2019-02-21:
You should be able to run doxygen in the root of Bullet to generate API level documentation.
You can see class hierarchies visually, for example [btCollisionObject](https://pybullet.org/Bullet/BulletFull/classbtCollisionObject.html
). 

Since moving to a new server last year, the old wiki hasn't been recovered, it has some useful bits in there.

I suggest learning from the examples in the BulletExampleBrowser, checking out the [BulletUserManual](https://github.com/bulletphysics/bullet3/raw/master/docs/Bullet_User_Manual.pdf) and browser the[ API info](https://pybullet.org/Bullet/BulletFull/index.html) and check out the [PyBullet Quickstart Guide](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.2ye70wns7io3).