bulletphysics/bullet3 #1066: pybullet: How to get all names of all links

benelot, 2017-04-10:
I am trying to a dictionary mapping from the name of a link to an object holding the link. The link class I already have (it basically runs methods of pybullet and stores the bodyid and the linkIndex of the link). However, I can not get proper names of the links using getBodyInfo() for all the bodyId, linkIndex pairs. How can I do this? If it does not exist, I will help to implement it in pybullet as it is very handy if you want to access links known to you by name.

erwincoumans, 2017-04-10:
See getJointInfo in the pybullet quickstart guide (http://pybullet.org), it has names

benelot, 2017-04-10:
Yes, for joints, but not for links. If you load a MJCF, there are link names (<body name="torso" ...) and joint names. (<joint name="rot2"...) I already found the joint names to access the joints via dictionary. But I need the link names too. That is pretty standard practice in MuJoCo to use the names.

benelot, 2017-04-25:
How would you like to make the link names accessible? I would suggest to introduce a getLinkInfo(bodyIndex, linkIndex) which could return all sorts of things like shapetype, scaling and my highly desired link name.

benelot, 2017-04-25:
Alternatively I could just append it to the jointInfo as another field. I just did this in a separate branch for my purposes since the linkName is already properly stored, but not accessible.

benelot, 2017-04-25:
Just give me a hint and I will make a PR.

closed 2017-04-26 by benelot