bulletphysics/bullet3 #1074: Issues found by PVS-Studio

skhrenov, 2017-04-17:
Hello,
I have found a few bugs using PVS-Studio analyzer. PVS-Studio is a static code analyzer for C, C++ and C#: [https://www.viva64.com/en/pvs-studio/](https://www.viva64.com/en/pvs-studio/)

Analyzer warnings:

- [V773](https://www.viva64.com/en/w/V773/) The function was exited without releasing the 'memoryBuffer' pointer. A memory leak is possible. pybullet loadmeshfromstl.h 49
[A link to the source code.](https://github.com/bulletphysics/bullet3/blob/874d02d7aac4a562d3ef36d654e607bfbb372e68/examples/Importers/ImportSTLDemo/LoadMeshFromSTL.h#L49)

- [V773](https://www.viva64.com/en/w/V773/) The function was exited without releasing the 'tga_data' pointer. A memory leak is possible. pybullet stb_image.cpp 2969
[A link to the source code.](https://github.com/bulletphysics/bullet3/blob/874d02d7aac4a562d3ef36d654e607bfbb372e68/examples/ThirdPartyLibs/stb_image/stb_image.cpp#L2969)

- [V522](https://www.viva64.com/en/w/V522/) Dereferencing of the null pointer 'mbl' might take place. pybullet physicsservercommandprocessor.cpp 379
[A link to the source code.](https://github.com/bulletphysics/bullet3/blob/874d02d7aac4a562d3ef36d654e607bfbb372e68/examples/SharedMemory/PhysicsServerCommandProcessor.cpp#L379)

In addition, I suggest having a look at the emails, sent from @pvs-studio.com.

Best regards,
Sergey Hrenov,
PVS-Studio Team

erwincoumans, 2017-04-17:
Thanks for the review, the 1st and 3rd are valid issues we need to fix. The 2nd seems pointless, since we check for 0 pointer, so there is nothing to be leaked.

erwincoumans, 2017-04-17:
Both issues are address in this commit:
https://github.com/bulletphysics/bullet3/commit/ad2099e107174759cf6d4d80469fd6c157858b89
Since the 2nd one isn't an issue, no action is needed.
Thanks!

ValZapod, 2019-07-30:
@erwincoumans Hello! There is another pvs-studio check from authors of PVS-studio... Please check it!😁😁 https://habr.com/ru/company/pvs-studio/blog/461841/

For example, this one https://github.com/bulletphysics/bullet3/blob/d0880a78d5d4b6146475a420b1b875eb91a62314/examples/SharedMemory/PhysicsClientC_API.cpp#L4145
Pi is wrong))

erwincoumans, 2019-07-31:
Thanks for the report! Most of the issues were in examples, which are less strictly developed.
Still, good to fix it, now it uses B3_PI:
https://github.com/bulletphysics/bullet3/blob/master/examples/SharedMemory/PhysicsClientC_API.cpp#L4145

Oh, and Bullet and PyBullet are still very actively developed and used, in particular for robotics and reinforcement learning.

erwincoumans, 2019-07-31:
We should probably get rid of BussIK and also from the Bullet3/OpenCL code, it is not used.

erwincoumans, 2019-07-31:
tinyrenderer example also fixed, for create cube x,y,z (and not x,y,y)

ValZapod, 2019-07-31:
@erwincoumans I did not wrote down all found moments... Can you please read it in https://habr.com/ru/company/pvs-studio/blog/461841/ (from 
"Over the top: exceeding array boundaries" to "Errors found before us")
For example, https://github.com/bulletphysics/bullet3/blob/cdd56e46411527772711da5357c856a90ad9ea67/examples/Importers/ImportBullet/SerializeSetup.cpp#L51

this pointer's memory was not released.

erwincoumans, 2019-07-31:
I'll check it out. Until then, don't restart that example more than a trillion times :)

ValZapod, 2019-08-03:
@erwincoumans I updated previous comment with my errors from my own PVS-Studio log. Wow__)) I will attach it. It is like 964 problems in it)) The second file is just for fun)) I mean it is only 3rd level warning (+1275). Usually not very interesting but you can look for it as well)) Use keywords: "memory" and "twice" and always "false" always "true"))
[project.log](https://github.com/bulletphysics/bullet3/files/3463787/project.log)
[project3.log](https://github.com/bulletphysics/bullet3/files/3463788/project3.log)

erwincoumans, 2019-08-03:
Is there a way to exclude examples, in particular examples/ThirdPartyLibs? and exclude Bullet/src/Bullet3* ?

ValZapod, 2019-08-03:
Yeah. For example, I should've excluded cmake folders. But they are in alphabet order in .log files...

closed 2021-04-26 by erwincoumans